eyalroz/printf #32: removal of out_fct_wrap_type

mickjc750, 2021-08-29:
This wrapper exists, as a means of passing a function pointer and a null pointer, from the interface function all the way down to the character output function, which itself is a function pointer based on the output mode. This is needed to provide fctprintf().

A better approach, is to instantiate a variable set (struct) particular to the output mode in the interface function. Then pass this by reference through to the character output function.
 This makes the character output function the same format (void (*out)(char, void*))  as that used by fctprintf().
This removes the need for a double function pointer de-reference per character which is a performance boost.
It also removes the need to pass the 'maxlen' parameter to most of the functions, where for most output modes it is unused.

eyalroz, 2021-09-06:
I don't exactly follow what you're suggesting:

* `out_fct_wrap_type` is not exposed in the library interface.
* As for the double-dereferencing - we do currently have it, but I don't see how your suggestion would remove it: You seem to be suggesting we pass the function pointer through a pointer to a struct containing it.

Please clarify...

mickjc750, 2021-09-07:
Not the function pointer. Only variables specific to the mode of output. It's difficult for me to explain, but I've already done this on an open PR, you can see the changes there.

eyalroz, 2021-09-07:
Oh, now I see your PR.

But - it's a 44-commit PR; that doesn't make sense. I can't use it without spending hours, or days, on merging. To submit a PR, you start from the branch you're PR'ing against, and only apply one or a few changesets to get the effect you wanted.

Anyway, I don't mind using the struct you suggested, which basically a kind of a richer iterator than a raw pointer. I'll try to manually merge that change soon.

eyalroz, 2021-09-07:
Hmm. This is interesting. This actually gives me the idea of doing something a little differentr:

* _Do_ wrap output functions.
* Use the wrapper for regular buffer output as well
* Have the wrapper take care of counting the number of characters printed
* Don't pass maxlen around at all... perhaps we can even avoid passing idx around entirely.
* Don't pass idx and maxlen to the output function - because they're supposed to have it.

eyalroz, 2021-09-26:
Well, I was going to work on this, but then other issues came up. Now that there are no open bugs, I believe it might be a good idea to pause, make a release (v5.0 I suppose), then get back to this. What do you think, @mickjc750 ?

mickjc750, 2021-09-27:
I agree, I have some other issues myself at the moment. It's working well enough for now.

eyalroz, 2022-01-22:
@mickjc750 : So, this is happening... expect a PR soon. It will do away with 4 parameters + the return value in lots of functions :-)

commit acf98a35 (2022-01-22): Fixes #32: Now using a convenient uniform gadget for the inner implementation for all library functions.

* Added: An `output_gadget_t` struct
* Added: Named constructor idioms for `output_gadget_t`, instead of the function wrappers we have before: For `putchar_()` output, buffer output, discarding/no output, and arbitrary function output.
* Implementation functions no longer need to maintain and pass around the parameters `buffer`, `maxlen`, `idx` - these are part of the uniform output gadget
* Implementation functions no longer need to increase the position within the output (`idx)`, nor ret [... cut]
commit c519fd6c (2022-01-22): Fixes #32: Now using a convenient uniform gadget for the inner implementation for all library functions.

* Added: An `output_gadget_t` struct
* Added: Named constructor idioms for `output_gadget_t`, instead of the function wrappers we have before: For `putchar_()` output, buffer output, discarding/no output, and arbitrary function output.
* Implementation functions no longer need to maintain and pass around the parameters `buffer`, `maxlen`, `idx` - these are part of the uniform output gadget
* Implementation functions no longer need to increase the position within the output (`idx)`, nor ret [... cut]
commit ae6b6991 (2022-01-24): Fixes #32: Now using a convenient uniform gadget for the inner implementation for all library functions.

* Added: An `output_gadget_t` struct
* Added: Named constructor idioms for `output_gadget_t`, instead of the function wrappers we have before: For `putchar_()` output, buffer output, discarding/no output, and arbitrary function output.
* Implementation functions no longer need to maintain and pass around the parameters `buffer`, `maxlen`, `idx` - these are part of the uniform output gadget
* Implementation functions no longer need to increase the position within the output (`idx)`, nor ret [... cut]
commit d60e53b9 (2022-01-25): Fixes #32: Now using a convenient uniform gadget for the inner implementation for all library functions.

* Added: An `output_gadget_t` struct
* Added: Named constructor idioms for `output_gadget_t`, instead of the function wrappers we have before: For `putchar_()` output, buffer output, discarding/no output, and arbitrary function output.
* Implementation functions no longer need to maintain and pass around the parameters `buffer`, `maxlen`, `idx` - these are part of the uniform output gadget
* Implementation functions no longer need to increase the position within the output (`idx)`, nor ret [... cut]
closed 2022-02-28 by eyalroz