eyalroz/printf #33: Feature request - runtime error trapping for disabled type specifiers

mickjc750, 2021-08-29:
Example:
%f is not available due to disabling PRINTF_SUPPORT_FLOAT_SPECIFIERS, and the application encounters a %f in a format string.
It would be useful to trap this so that the application can cause a runtime fault with an assertion handler.
This could be done with a macro which is ((void)0) by default, but could be edited in the config file to be something like ASSERT(false)

eyalroz, 2021-09-06:
So, a three-valued option for specifiers: support, ignore, trap.

I can see why this might have some use in debugging printf-related code, but - is it really the library's business? Someone can simply parse the format string before passing it to the printf-like function.

mickjc750, 2021-09-07:
Not a 3 valued option, either they're supported or they're not. But, if they're not and one is encountered, printf_ could provide some mechanism for executing an error handler. I would do this with a macro, but since the _config file is only a build artifact, I'm not sure what the best mechanism would be in your fork.
 Is it the libraries business? If it's configured in a way that makes it misbehave in it's application, yes.

eyalroz, 2021-09-07:
> But, if they're not and one is encountered, printf_ could provide some mechanism for executing an error handler. 

Why? If printf'ing is important enough for its non-conformance to be an error you want to handle, then just enable all of its features and the issue goes away.

> Is it the libraries business? If it's configured in a way that makes it misbehave in it's application, yes.

* It will behave the way it's configured to behave. 
* Why should unsupported %f of %e behave differently than %w or %j or any other invalid format specifier? Silently ignoring these is the way full-blown `printf()` behaves.
* If someone wants to make sure the library is not sent unsupported/invalid format specifier, they can just write a wrapper which checks the format string. They don't need library support for that.

eyalroz, 2021-09-22:
So, not convinced about the merit of this. If more users ask for it, I might reconsidered. Closing for now.

closed 2021-09-22 by eyalroz