eyalroz/printf #34: Feature - request, runtime warning trap for overrun of PRINTF_FTOA_BUFFER_SIZE or PRINTF_NTOA_BUFFER_SIZE

mickjc750, 2021-08-29:
It would be useful to trap the event where the output becomes affected by the configured buffer sizes PRINTF_FTOA_BUFFER_SIZE or PRINTF_NTOA_BUFFER_SIZE.

This could be a macro in the printf_config.h which by default is ((void)0), but can be edited to something like WARN(false), depending on the applications runtime warning handler.

eyalroz, 2021-09-06:
Not sure it's a good idea to clutter the code with something like this, especially since it will need either changes in many places in the code. Convince me this is both important will be used by many?

mickjc750, 2021-09-07:
Consider the potential consequences of this. Not all readers of text are human. The application may be generating AT commands for a modem, or textual commands for some other module. Or, maybe it's producing a .csv or text based file which is then read by other software. The symptom of a malfunction may only occur in rare cases, and it may take a long time for the developer to track the problem all the way back to the printf_ configuration. 
 Will it be used by many? I can't speak for other people, but I believe my reasons for wanting a runtime warning on this are good ones.
 It may be easier to implement, if the warning occurred when the buffer size was reached, but not necessarily exceeded. For NTOA, this only needs a comparison between len & the buffer size at the end of _ntoa_format(). As for FTOA, I'm yet to work that out...

eyalroz, 2021-09-07:
But if this is an issue, i.e. when printf'ing is mission-critical rather than just a debugging why would one not simply set the buffers to  an appropriate size?

diggit, 2021-09-07:
There could be bugs (insufficient buffer size, etc) even in mission critical code. Everyone makes mistakes.
Imo regular `assert` is the right tool to catch such mistakes. Asserts get disabled in release builds by defining `NDEBUG`.
You culd wrap it in another layer macros to get printf specific switch disabling assert checks.

eyalroz, 2021-09-22:
@diggit : Fair enough; would you like to submit a PR with the relevant locations for assertions? (The alternative is to wait for me to get around to doing it.)

eyalroz, 2021-11-11:
@diggit : ping.

diggit, 2021-11-11:
I was just giving my opinion about the right way. I'd have to study whole codebase and that's not feasible ATM from my side.

closed 2021-11-11 by mickjc750