eyalroz/printf #35: Build warning when compiled with avr-gcc (GCC) 5.4.0

mickjc750, 2021-08-29:
printf.c: In function ‘sprint_exponential_number’:
	printf.c:491:25: warning: ‘normalization.raw_factor’ may be used uninitialized in this function [-Wmaybe-uninitialized]
	   struct scaling_factor account_for_precision = update_normalization(normalization, prec_power_of_10);
                         ^
	printf.c:625:25: note: ‘normalization.raw_factor’ was declared here
	   struct scaling_factor normalization;

eyalroz, 2021-09-05:
If you add 
```
should_skip_normalization = true;
```
within the ` if (abs_number == 0.0) {` block (line 629), does that make the warning go away?

mickjc750, 2021-09-06:
No, the warning remains.

 The if block at 630-658 will either assign exp10=0, or assign a value to normalization.raw_factor. With exp10==0 the should_skip_normalization boolean will be true, and get_normalized_components() will not be executed.

So the warning is not a real threat. I propose we just add an initializer to the declaration of  'normalization' on line 625.

mickjc750, 2021-09-06:
No, that produces an error, should_skip_normalization is undeclared.
If I fix that error by declaring should_skip_normalization at the start of
the function (instead of line 683), the warning still remains.

Do we know if the warning is a real threat? maybe we just init on line 625
in it's declaration?

Regards,

-- 
--
Michael Clift - Design Engineer.

Clift Innovations Pty Ltd
Suit 10/13 Walkers Road
Nunawading, VIC 3131

ABN 24 286188328

Tel: +61 3 9877 3109
Fax: +61 3 9999 1406
Mob: 0417 613 680
Email: ***@***.***
Web: www.clift.com.au
*****************************************************************************
PRIVATE AND CONFIDENTIAL
This e-mail and any files transmitted with it are intended solely for the
use of the addressee(s) and may contain information that is confidential or
privileged. If you receive this e-mail and you are not the addressee(s),
please disregard the contents of the e-mail, delete the e-mail and notify
the author immediately.
*****************************************************************************

eyalroz, 2021-09-06:
Michael, please don't reply with a huge quotation block, that's useless for these bug pages. Just your reply, and nothing else please...

eyalroz, 2021-09-22:
@mickjc750 : ping.

eyalroz, 2021-09-25:
So, this is a compiler bug. The value cannot be used uninitialized. MSVC also fails to realize this.

Closing for now.

closed 2021-09-25 by eyalroz