eyalroz/printf #38: If %e specifiers are enabled without enabling %f specifiers, then neither are provided.

mickjc750, 2021-08-29:
Enabling option PRINTF_SUPPORT_EXPONENTIAL_SPECIFIERS but not PRINTF_SUPPORT_FLOAT_SPECIFIERS.
Yields neither..
This may occur if an application really only wants the exponent format.

eyalroz, 2021-09-06:
I don't actually see this happening...
```
/home/eyalroz/src/mine/mpaland-printf/test/test_suite.cpp:849: PASSED:
  CHECK( !strcmp(buffer, "4.895512e+04") )
with expansion:
  true
with messages:
  test::sprintf_ arguments, replicated ( "arg := arg" ):
  ----
  "%e" := "%e"
  48955.125 := 48955.125
  ----
  Resulting buffer contents: "4.895512e+04"
```
etc.

eyalroz, 2021-09-18:
@mickjc750 : ping.

mickjc750, 2021-09-18:
@eyalroz 
I'm not sure how to reproduce your snippet from the test_suit. I originally saw this issue when compiling the .c and .h files alongside the AVR test. But I can reproduce it on a desktop with the following steps :

Edit the printf_config.h.in to the following:
 #cmakedefine PRINTF_SUPPORT_DECIMAL_SPECIFIERS       0
 #cmakedefine PRINTF_SUPPORT_EXPONENTIAL_SPECIFIERS   1
 #cmakedefine PRINTF_SUPPORT_LONG_LONG                1
 #cmakedefine PRINTF_ALIAS_STANDARD_FUNCTION_NAMES    1

Edit the autotest.cpp to include the following on line 118 after the usage message:
    sprintf_(txt, "%%e = %e %%f = %f", 8E41, 0.97341);  
    printf("%s\n",txt);

Then:
 cmake .
 make
 sudo make install
 cd test
 cmake .
 cp ../include/printf_config.h .
 make
 ./autotest

The output after the usage message shows:
%e = e %f = f

eyalroz, 2021-09-19:
Well, it seems I have a bug with the CMake generation of `printf_config.h`. When I fix that, I see what you're seeing.

eyalroz, 2021-09-19:
@mickjc750 : Please try this out.

mickjc750, 2021-09-20:
@eyalroz confirmed fixed.

commit 81f12973 (2021-09-19): Fixes #38: No longer rejecting exponential specifiers when only decimal specifiers are disabled. Also, corrected some test suite code which caused failures when only decimal or only exponential specifiers are enabled.
closed 2021-10-05 by eyalroz