eyalroz/printf #39: Build warning regarding sprint_exponential_number() when %ll and %f are enabled without %e

mickjc750, 2021-08-29:
The following build warning occurs with:
PRINTF_SUPPORT_FLOAT_SPECIFIERS   ON
PRINTF_SUPPORT_EXPONENTIAL_SPECIFIERS    OFF
PRINTF_SUPPORT_LONG_LONG     ON

Built with arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

app/src/printf.c: In function 'sprint_floating_point':
app/src/printf.c:779:10: warning: implicit declaration of function 'sprint_exponential_number'; did you mean 'sprint_decimal_number'? [-Wimplicit-function-declaration]
          sprint_exponential_number(out, buffer, idx, maxlen, value, precision, width, flags, buf, len) :
          ^~~~~~~~~~~~~~~~~~~~~~~~~
          sprint_decimal_number

eyalroz, 2021-09-06:
How this?

mickjc750, 2021-09-18:
@eyalroz yes that fixes it.

eyalroz, 2021-09-19:
The issue will get auto-closed when the chage lands on the main branch...

commit 4123dc00 (2021-09-06): Fixes #39: #ifdef-out `sprint_exponential_number()` when the exponential specifiers are disabled.
commit 37bdb950 (2021-09-27): Fixes #39: #ifdef-out `sprint_exponential_number()` when the exponential specifiers are disabled.
closed 2021-10-05 by eyalroz