eyalroz/printf #54: Alias preprocessor define is working?

thestumbler, 2021-11-08:
I tried to use the alias function names define, and as best I can tell it didn't do anything. Looking at the code, it isn't clear to me how it's supposed to work. I am on the latest 27 Sep commit. 

```
#ifdef PRINTF_ALIAS_STANDARD_FUNCTION_NAMES
# define printf_    printf_
# define sprintf_   sprintf_
# define vsprintf_  vsprintf_
# define snprintf_  snprintf_
# define vsnprintf_ vsnprintf_
# define vprintf_   vprintf_
#endif

// ... some stuff here ...

#ifdef PRINTF_ALIAS_STANDARD_FUNCTION_NAMES
# undef printf_
# undef sprintf_
# undef vsprintf_
# undef snprintf_
# undef vsnprintf_
# undef vprintf_
#endif
```

I temporarily hand-edited my local `printf.h` file and went back to way function name aliasing worked in the previous 17 Aug commit. This works, and I understand why it works (unlike the current commit).

```
#ifdef PRINTF_ALIAS_STANDARD_FUNCTION_NAMES
# define printf    printf_
# define sprintf   sprintf_
# define vsprintf  vsprintf_
# define snprintf  snprintf_
# define vsnprintf vsnprintf_
# define vprintf   vprintf_
#endif

// ... some stuff here ...

```
I did read the issue #14 notes, but they didn't explain what I'm experiencing. Regarding the 27 Sep commit, am I missing something about how to make it work and/or understanding how it should work? 

Thanks,
-Chris

eyalroz, 2021-11-08:
Hmm. I think I might have gotten that wrong. In the recent commit, change the first set of directives to:
```
#ifdef PRINTF_ALIAS_STANDARD_FUNCTION_NAMES
# define printf_    printf
# define sprintf_   sprintf
# define vsprintf_  vsprintf
# define snprintf_  snprintf
# define vsnprintf_ vsnprintf
# define vprintf_   vprintf
#endif
```
and that should do it.

eyalroz, 2021-11-08:
I did mess that up. Working on it.

eyalroz, 2021-11-11:
@thestumbler : Can you please try it again now?

thestumbler, 2021-11-12:
I just tested it, the latest commit on the develop branch. No joy.
First of all , my compiler (gnu arm-none-eabi-g++ v 6.3.1 Atmel build for Atmel/Microchip SAM-E5x chip) doesn't like the preprocessor #if statement without a conditional (the second line from the snippet below)

```
#ifdef PRINTF_ALIAS_STANDARD_FUNCTION_NAMES
#if PRINTF_ALIAS_STANDARD_FUNCTION_NAMES
# define printf_    printf
# define sprintf_   sprintf
# define vsprintf_  vsprintf
# define snprintf_  snprintf
# define vsnprintf_ vsnprintf
# define vprintf_   vprintf
#endif
#endif
```

Even when commenting this out in all three places (twice in the h file and once in the c file), it doesn't actually work.
I got it to work by basically putting it back like before:

(1) swapped all of the definitions around at the top of the header file, e.g...

```
WAS: # define printf_    printf
 IS: # define printf    printf_
```
(2) removed the #undef block at the end of the header file, and removed the duplicate? block at the top of the c file.

thestumbler, 2021-11-12:
"commenting this out in all three places..." I mean like this:

```
#ifdef PRINTF_ALIAS_STANDARD_FUNCTION_NAMES
//#if PRINTF_ALIAS_STANDARD_FUNCTION_NAMES
# define printf_    printf
# define sprintf_   sprintf
# define vsprintf_  vsprintf
# define snprintf_  snprintf
# define vsnprintf_ vsnprintf
# define vprintf_   vprintf//
//#endif
#endif
```

eyalroz, 2021-11-12:
> First of all , my compiler (gnu arm-none-eabi-g++ v 6.3.1 Atmel build for Atmel/Microchip SAM-E5x chip) doesn't like the preprocessor #if statement without a conditional (the second line from the snippet below)

Ok, I'll change it to
```
#ifndef PRINTF_ALIAS_STANDARD_FUNCTION_NAMES
#define PRINTF_ALIAS_STANDARD_FUNCTION_NAMES 0
#endif
```

> Even when commenting this out in all three places (twice in the h file and once in the c file), it doesn't actually work. I got it to work by basically putting it back like before:

The thing is, I can't do that. We can't go around replacing `print`'s in people's source code, outside the library, with `printf_`. Those replacements will happen in comments, in C++ namespaces... can't have that. The macros must be undefined at the end of the header.

commit d5803c5c (2021-11-08): Fixes #54: Fixed the mess I created in the not-really-a-fix for issue #14. Added a test for whether aliasing works, in which the standard library function names are used, without including the `<stdio.h>` or any other standard library headers.
commit c3a95cd3 (2021-11-08): Fixes #54: Fixed the mess I created in the not-really-a-fix for issue #14. Added a test for whether aliasing works, in which the standard library function names are used, without including the `<stdio.h>` or any other standard library headers.
closed 2021-11-19 by eyalroz