eyalroz/printf #58: Should I move printf.c and printf.h into a subdirectory?

eyalroz, 2021-11-11:
The repository root has a bunch of non-source files in addition to `printf.c` and `printf.h`; this is slightly distracting and messy. In larger repositories, one typically places the sources files in (one or more) subdirectories, and none of them in the repository root.

Should we, then, create a `src`/`source` subdirectory, and movie `printf.h` and `printf.c` into it?

diggit, 2021-11-11:
Larger projects usually have struture like `src/printf.c` and `inc/projectName/printf.h`. This approach allows adding of `somePath/LocalSubmoduleName/inc` into include path. Then you include headers with projectName path like `#include "printf/printf.h" `. Such system minimizes possibility of filename collisions.

But it's probably beneficial only if you want to separate headers from sources, because otherwise, header prom same directory has highest inclusion priority and there is no issue if user has some other `printf.h` in their include path.

eyalroz, 2021-11-11:
> Larger projects usually have struture like `src/printf.c` and `inc/projectName/printf.h`. 

Some put everything in `src/`, some separate an `include/` directory. Personally, and since I separate the repo structure from the installed directory structure, I like them both under the same subdir.

> This approach allows adding of `somePath/LocalSubmoduleName/inc` into include path. 

But that's not my suggested use. If you build the library with CMake, then install it (for example, under `/opt/printf`, you get:
```
/opt/printf/
  \-- include
     \-- printf.h
  \-- lib
     \-- libprintf.so
     \-- cmake
        \-- printf
           \-- printf-config-version.cmake
           \-- printf-config.cmake
           \-- printf-config-release.cmake
```
and the CMake files set up the targets for you, so you don't get any collisions. If you are worried about collisions, you can install under `/opt/printf/printf`, and then use `-I/opt/printf` with `#include <printf/printf.h>`.

closed 2021-11-19 by eyalroz

edubart, 2021-12-29:
Both files could be merged into one, turning the library in a single header library, like `nanoprintf` and `stb_sprintf`. I think It's more easy to use single header C libraries.

eyalroz, 2021-12-29:
@edubart : Please open a separate issue for that request.

commit cbff122b (2021-11-11): Resolves #58: Moving `printf.h` and `printf.c` into the `src/` directory (and marking it as an include directory).
commit 78eaa13a (2021-11-12): Resolves #58: Moving `printf.h` and `printf.c` into the `src/` directory (and marking it as an include directory).
commit 0ea44381 (2021-11-12): Resolves #58: Moving `printf.h` and `printf.c` into the `src/` directory (and marking it as an include directory).