eyalroz/printf #69: question re: my compiler tuple supports 32b ints only . . . out of luck?

toddcromiii, 2021-12-06:
Looking over the code, I noted that there is apparent support of 32b doubles (via DOUBLE_SIZE_IN_BITS and friends), and avoid making use of (u)int64_t in that case. But: my compiler supports only 32b ints. Does that mean I am out of luck if I want to define PRINTF_SUPPORT_DECIMAL_SPECIFIERS 
 or PRINTF_SUPPORT_EXPONENTIAL_SPECIFIERS? The code makes use of int_fast64_t, which size I can't do. Has this issue been discussed? (I'd mark this as a question but do not know how)

eyalroz, 2021-12-07:
@toddcromiii : This issue has not been discussed, or at least - not discussed in context of this repository. The reason is that C99 supposedly offers 64-bit integer types, so naturally it is assumed they are available. Does your compiler not offer emulation for 64-bit types using pairs of 32-bit types?

toddcromiii, 2021-12-07:
Thank you! It helps to know that. The compiler is an experimental port of llvm/clang to ez80 (not by me). BTW in my view llvm/clang is a beast, and I don't mean that enthusiastically, but the Zilog C is rife with internal errors (and definitely doesn't support 64b ints).

So far I believe the support for 64b ints in the llvm/clang port is incomplete. I have not ascertained how incomplete; the compiler allows long long but last time I tested it, simple math operations on the type did not yield correct results. I will have to look into that further. Also interesting is the emulation possibility you alluded to; if llvm/clang allows this generally, that might be a way to solve the problem. I doubt these are going to bear fruit, tho, so as the printf code here stands, I currently believe I am indeed out of luck. If it weren't smack in the middle of the floating point stuff, I could probably fix the code readily, but I am not very expert at floating point, so it'd take some study and work. As it stands, if you like, as far as I am concerned, you can close the issue, as I was seeking a quick answer as to the general state and you very helpfully provided that!

eyalroz, 2021-12-07:
>  the compiler allows long long but last time I tested it, simple math operations on the type did not yield correct results.

Bear in mind that `long long int` is not required to have 64 bits. It's only required to be no smaller than `long int`, which in turn is no smaller than `int` etc.

eyalroz, 2021-12-07:
Anyway, it is possible to alter the code of this library so as not to rely on 64-bit integers - and _without_ going into floating point intricacies. You would need to carefully follow the use of `double_uint_t`, and replace it with a structure of two 32-bit integers - especially as the `U` field of `double_with_bit_access_t`, and in the decimal_components structure. Your type conversions and arithmetic will naturally be more complicated, but this is quite doable. It will become even easier if your `double` size happens to be 32-bits as well.

toddcromiii, 2021-12-07:
Thank you! Yeah, I forgot to mention that yeah, I know about the relational size guarantees re ints, but that long long int is a fledgling and incomplete attempt at, yes, 64b on this compiler. The idea of using two 32b ints and implementing arithmetic, shifting, and well, it looked like there was some multiplication too, is a possible avenue . . . just a bit involved. I did make a mult 24b by 8b function in ez80 assembly using the ez80 8b x 8b multiply, so I have some experience with piecing together multiplication (and division as well, tho my division fn is not optimal), but it wasn't for the faint of heart and I was able to test all possible multiplications of 24b x 8b brute forcedly to further convince myself, but add wider types and use further to implement floating point, and that's not an afternoon's work any more for me (tho it is cool there is a test suite) :)

toddcromiii, 2021-12-10:
Hmm, just in case anyone else looks at this thread: looking at print_floating_point and functions below it and types used, through some careful choices of macros I may be able to use uint32_t for the type currently written as int_fast64_t as used in double_components (as well as a couple other places).

Such a path may be simpler than trying to cobble together 64b arithmetic for not much gain. They already take pains to "limit precision so that our integer holding the fractional part does not overflow" in this fn, and I think some lines above that they may avoid overflowing for the "integral" part of double_components. Of course one cannot hold values above 18 quintillion or whatever in the integral part, so they presumably have to draw the line somewhere in any case (plus avoid the straight cast from a double value (abs_number) to the "integral" part (in get_components) from overflowing even 64b). So through careful choices of some macros like  NUM_DECIMAL_DIGITS_IN_INT64_T and friends, I may be able to get it working more simply than the earlier path we discussed. Of course, convincing myself it works is another matter (but that would appear as a hurdle in either path).

eyalroz, 2021-12-10:
Shall I close this issue now? Or - are you planning to submit a PR at some point?

toddcromiii, 2021-12-10:
Thank you! Closing it is fine. If I came up with something that I felt was tested or otherwise confidence-making enough for others to use, we could address that then.

closed 2021-12-10 by toddcromiii