eyalroz/printf #70: Need of CI server?

IvanVnucec, 2021-12-07:
Hey, 
I see that there is no CI server implemented at the moment. Maybe this project needs one? If so, I'm willing to help you out? I have experience with the GitHub Actions (see my repos where I use Actions extensively [MYREPO1](https://github.com/IvanVnucec/c_matrix_library), [MYREPO2](https://github.com/IvanVnucec/Optimal-REQUEST)).

eyalroz, 2021-12-07:
> maybe this project needs one? 

Well, I doubt it needs one strictly for building, as after all this project has no dependencies and uses plain vanilla C99, so any C compiler will do too. However, it would still be nice to have this.

Actually - if you could have the tests run as well (and make sure that they pass) using GitHub actions, that would be nice. I myself am not very proficient with GitHub Actions and am still gritting my teeth with them in [another project of mine](https://github.com/eyalroz/cuda-api-wrappers/) - so I would definitely welcome a PR.

IvanVnucec, 2021-12-07:
Ok, I will fork this project and make the following changes:
- GH Actions server running Ubuntu (version to be defined) and
- building with CMake and GCC compiler (version to be defined, maybe multiple ones idk) on every commit and pull req and
- running tests and
- add build and test report artifacts and
- add Code coverage report and
- add Build and Code coverage badges to the README 

If you want to add or exclude something, let me know.

eyalroz, 2021-12-07:
Depending on how much time/interest in this you have, I would add:

* Additional build platforms: Windows, MacOS, Some kind of BSD.
* (This one is tricky) cross-build for some relevant embedded platform (would be nice to try one which is limited to 32 bits or less).

Also note the "autotest" fuzzer. It essentially runs indefinitely, printing mismatches as it finds them. It's not in the CMake test suite, so you can ignore it for now, but theoretically it could be

Finally, regarding the GCC version - it doesn't really matter to me at this point. If you really like, you could try building with multiple compilers and versions.

I am the most thrilled about the code coverage report, because that would motivate me to add additional testcases.

IvanVnucec, 2021-12-07:
Ok, I will add your suggestions to the list of things we want.
Regarding "autotest", I will ignore it once we have successfully implemented a working CI server.

eyalroz, 2022-02-20:
If you want to help improve what's in place right now for this repo, you're still welcome.

closed 2022-03-04 by eyalroz