eyalroz/printf #74: (minor) Some functions missing prototypes

fenugrec, 2021-12-12:
Compiling with -Werror=missing-prototypes (I have that set globally in a project), causes these:

```
gpibusb-firmware-arm/printf/src/printf/printf.c:483:8: error: no previous prototype for 'apply_scaling' [-Werror=missing-prototypes]
  483 | double apply_scaling(double num, struct scaling_factor normalization)
      |        ^~~~~~~~~~~~~
gpibusb-firmware-arm/printf/src/printf/printf.c:488:8: error: no previous prototype for 'unapply_scaling' [-Werror=missing-prototypes]
  488 | double unapply_scaling(double normalized, struct scaling_factor normalization)
      |        ^~~~~~~~~~~~~~~
gpibusb-firmware-arm/printf/src/printf/printf.c:493:23: error: no previous prototype for 'update_normalization' [-Werror=missing-prototypes]
  493 | struct scaling_factor update_normalization(struct scaling_factor sf, double extra_multiplicative_factor)
      |                       ^~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors

```

Perhaps those functions could be declared as static since (if I understand correctly) they're not meant to be used outside of printf.c ?

eyalroz, 2021-12-12:
Sure, why not. Note this won't reach the master branch for a while, as I've just released 5.2.0.

fenugrec, 2021-12-12:
that's fine, I'm pulling in as a git submodule here. Thanks !

phillipjohnston, 2022-01-25:
I saw this tagged as fixed-on-develop, but I get this warning in GCC-9 on the develop branch (none on master):

```
../printf/src/printf/printf.c: At top level:
../printf/src/printf/printf.c:952:16: warning: no previous declaration for 'parse_flags' [-Wmissing-declarations]
  952 | printf_flags_t parse_flags(const char** format)
      |
```

eyalroz, 2022-01-25:
@phillipjohnston : Regressed by a recent commit...

eyalroz, 2022-01-25:
@phillipjohnston : done.

commit 0267db61 (2021-12-12): Fixes #74: Fixes #74: Functions in `printf.c` without `printf.h` prototypes are now `static`, and will not raise any eyebrows when the resulting object is inspected.
commit e11965e1 (2022-01-05): Fixes #74: Fixes #74: Functions in `printf.c` without `printf.h` prototypes are now `static`, and will not raise any eyebrows when the resulting object is inspected.
commit 29a7423c (2022-01-26): Fixes #74: Fixes #74: Functions in `printf.c` without `printf.h` prototypes are now `static`, and will not raise any eyebrows when the resulting object is inspected.
closed 2022-02-28 by eyalroz