eyalroz/printf #76: printf(%c, x) : pulling in libg_nano stuff ?

fenugrec, 2021-12-16:
Hi,
I'm most likely something wrong here, I'm not sure...

```
#include <stdbool.h>
#include <string.h>

#include "printf_config.h"	//just to get PRINTF_ALIAS_STANDARD_FUNCTION_NAMES...
#include <printf/printf.h>

.........
			printf("eos_string: 0x");
			//printf("%c", eot_char);
```

I'm compiling this for embedded (ARM CM0) targets with nano.specs etc, and everything works as expected until I uncomment that second printf : then the linker tries to locate a bunch of libc stubs that are missing , e.g. _sbrk , _write_r, _seek_r etc (as I would expect if I was using the  stdio printf ). 
Is there anything special about using the "%c" specifier on its own ? If I replace that second printf with a %d ,  it'll work...

eyalroz, 2021-12-16:
> Is there anything special about using the "%c" specifier on its own ? 

Nope, nothing special about it. Also, as you know, the only things the library takes from the standard library are the definitions in:
```
#include <stdbool.h>
#include <stdint.h>
```
and that's that.

> If I replace that second printf with a %d , it'll work... Or if I change the formatter as " %c (i.e. adding a space in front), that also works fine !

Could it perhaps be some special optimization in your compiler? Maybe it replaces `printf("%c")` with `putchar()`?

fenugrec, 2021-12-16:
> Could it perhaps be some special optimization in your compiler? Maybe it replaces `printf("%c")` with `putchar()`?

10/10 . I just checked the disassembly of the .o for that module and that's exactly what is happening. Thanks !

Of course I can put putchar_() there myself but I'll try to find a "clean" way to do this. Maybe alias putchar() to putchar_ or something.

fenugrec, 2021-12-16:
Maybe... I notice it does the same thing with a printf() with no formatters, replacing it with puts()... usually gcc bug reports for embedded stuff like this falls on deaf ears though.

eyalroz, 2021-12-16:
@fenugrec : If you file it, I'll comment favorably and upvote :-)

Anyway, I know they take a long while to handle bugs, but then a few years later, it sometimes happens. Think of it is a s a civic duty.

eyalroz, 2022-01-25:
@kjsdf7wr3 : 

1. Which compilers support `-fno-builtin-printf`?
2. Which compiler were you using? The compiler should not do this if the source file has not, at the very least, included `stdio.h` .

fenugrec, 2022-01-25:
>     1. Which compilers support `-fno-builtin-printf`?

At least [gcc](https://gcc.gnu.org/onlinedocs/gcc/Other-Builtins.html) and [clang](https://clang.llvm.org/docs/ClangCommandLineReference.html)

I'm fairly certain I did not have an `#include <stdio.h>` in there. With gcc, I think this due to the following: 

> The ISO C90 functions [......] printf, putchar, puts, ..... are all recognized as built-in functions unless -fno-builtin is specified (or -fno-builtin-function is specified for an individual function).

Some other functions may/may not be recognized as builtins in certain  "strict ISO C modes"  (-ansi, -std=c90, -std=c99 or -std=c11) , I haven't explored this.

eyalroz, 2022-01-25:
@fenugrec , @kjsdf7wr3 : Please try pulling branch `fix-issue-76`, to check whether it resolves the problem. I'll also add a README.md note.

fenugrec, 2022-01-25:
Cannot test easily here (I'm not using the CMakeLists from this project, and just passing -fno-builtin-printf globally ) but your method seems reasonable; I assume it works on your setup ?

eyalroz, 2022-01-25:
@fenugrec : Well, it works in the sense that it builds and passes the tests. I haven't scrutinized the object code to check whether before the change I would get some putchar implementation pulled in.

kjsdf7wr3, 2022-01-26:
> @fenugrec , @kjsdf7wr3 : Please try pulling branch `fix-issue-76`, to check whether it resolves the problem. I'll also add a README.md note.

I use `printf.h` and `printf.c` in my code base directly, so I cannot comment on that. Maybe we should first come up with a minimal working example to trigger this feature.

eyalroz, 2022-01-26:
@kjsdf7wr3 :  If adding the switch works for you, I guess I'll just "trust myself" and go ahead and merge this; and if there's another complaint we can always reopen the issue.

kjsdf7wr3, 2022-01-26:
I forgot to say, that if you build printf as a library, adding the flag to the library build should have no effect on this issue.
The flag (`-fno-builtin-printf`) needs to be set where printf is called.

eyalroz, 2022-01-26:
> I forgot to say, that if you build printf as a library, adding the flag to the library build should have no effect on this issue. The flag (`-fno-builtin-printf`) needs to be set where printf is called.

Yes, but I used a PUBLIC compile option, which is propagated to targets depending on the library (if you use CMake properly).

commit 61d345a4 (2022-01-25): Fixes #76: Now preventing clang and gcc from replacing our aliased functions with strength-reduced actual standard library functions.
commit c4b8637c (2022-01-25): Fixes #76: Now preventing clang and gcc from replacing our aliased functions with strength-reduced actual standard library functions.
commit b1afc124 (2022-01-26): Fixes #76: Now preventing clang and gcc from replacing our aliased functions with strength-reduced actual standard library functions.
commit d789f4e6 (2022-01-31): Fixes #76: Now preventing clang and gcc from replacing our aliased functions with strength-reduced actual standard library functions.
closed 2022-02-28 by eyalroz