eyalroz/printf #77: MISRA compliance

sanderh255, 2021-12-24:
Critical embedded systems (medical and safety related) often require MISRA compliance. It would be great to implement this for the printf library.

eyalroz, 2021-12-24:
You'll need to be more specific about what that means.

sanderh255, 2021-12-24:
MISRA is a set of guidelines for safety critical code. It includes things like restricting the usage of dynamic memory allocation (which is one of the reasons this project is pretty cool), avoiding side-effect in conditional statements and using clear control flow (single return point in a function, and not using break to terminate a loop). These rules may or may not be to your taste, so I would like to implement them in a personal fork so you can decide whether or not this is something you'd like to include.

eyalroz, 2021-12-24:
> restricting the usage of dynamic memory allocation (which is one of the reasons this project is pretty cool)

Indeed, there is no dynamic memory allocation in libprintf.

> avoiding side-effect in conditional statements

You mean, ternary operator applications? I have some of these which do a `va_arg()`.

> These rules may or may not be to your taste, so I would like to implement them in a personal fork so you can decide whether or not this is something you'd like to include.

Ok, sure. Please try to avoid any other changes though. When you have a PR, I'll consider it. Is there some sort of MISRA verification you want to have run? You could perhaps write a GitHub action for that.

eyalroz, 2022-01-05:
So, I think this covers everything except the use of raw types.

Is there an online MISRA compliance checker, by the way?

sanderh255, 2022-01-05:
We use SonarQube (https://www.sonarqube.org/) which does have a free edition, but this edition does not include coverage for C/C++.
The only one I'm aware of is https://www.sonarlint.org/, which comes in the form of an IDE plugin.

eyalroz, 2022-01-05:
MISRA specs links (for myself...):

* [MISRA-C:2004](http://caxapa.ru/thumbs/468328/misra-c-2004.pdf)
* [MISRA-C:2012](http://my.ldrasoftware.co.uk/repository//miscellaneous/Misra-c_2012_compliance.pdf)

sanderh255, 2022-01-05:
It does seem the last commits somehow caused ```#if PRINTF_SUPPORT_EXPONENTIAL_SPECIFIERS``` to be duplicated instead of moved. It is now on line 417 and 513, but there does not seem to be a terminating ```#endif```.
Did you mean to remove the one on line 513?

eyalroz, 2022-01-05:
Umm... I don't see a duplicate `#if`... if it had existed, compilation should have failed.

sanderh255, 2022-01-05:
You're right, my bad. I seem to be unable to keep my fork up to date with your changes. Merging your changes into my develop did not do what I would expect it to.

eyalroz, 2022-01-05:
@sanderh255 : I'm partly to blame, since I often rewrite the history of the develop branch. Only master is safe in that respect. So when working with the develop branch here it's usually better to rebase -i develop , and remove the redundant commits from the todo list.

sanderh255, 2022-01-05:
I see, that makes sense. Thanks!

eyalroz, 2022-01-25:
So, I think we're done with this issue. If I've missed something, please reopen.

closed 2022-01-25 by eyalroz