eyalroz/printf #78: Convert to a single header library

edubart, 2021-12-29:
Would be nice if his library gets converted to a single header library. The ideia is to merge both `printf.c` into `printf.h` and then the functions are implemented only when defining `PRINTF_IMPLEMENTATION`.

Some examples of other single header printf libraries are:
https://github.com/charlesnicholson/nanoprintf/blob/main/nanoprintf.h
https://github.com/nothings/stb/blob/master/stb_sprintf.h

They could be used as examples on how to do it.  

This was mentioned in issue https://github.com/eyalroz/printf/issues/58

eyalroz, 2021-12-29:
Hmm. I suppose that, at the very least, there is some sort of a convenience benefit to having this library available as a single header. However, it definitely _should_ also be available as a header + implementation; and in that case, I do not want people to be faced with a 1,300-line file instead of a 150-line file.

What would you say to having a single header generated as part of (some sort of) a build process? Or do you believe it important to have one available apriori?

edubart, 2021-12-29:
> What would you say to having a single header generated as part of (some sort of) a build process? Or do you believe it important to have one available apriori?

If making this, I think it is important to have at least one single header file available for quick download in this repository. So we could grab the newest library downloading just one file from this repository, instead of having to clone the full repository and generate manually.
`printf.c` could still exist as just 2 lines:

```c
#define PRINTF_IMPLEMENTATION
#include "printf.h"
```

If you want to auto generate this, is your call. I think that to make a auto generating script may take a little extra work than just merging both files and adding a few ifdefs.

> I do not want people to be faced with a 1,300-line file instead of a 150-line file.

Usually I don't see much problem, a large header does not impact compile speed if the implementation is disabled. Personally I find easier to navigate/edit/maintain the code and its details when all the code is in a single file.

eyalroz, 2021-12-29:
> If making this, I think it is important to have at least one single header file available for quick download in this repository. 

How about as a release artifact?

eyalroz, 2022-01-05:
I am mulling this over... in the mean time a few other minor changes are being committed.

JakeSays, 2022-01-07:
I am really against this. You'll be adding more messy #ifdef's to a clean code base. I use this library extensively and it works great just the way it is.

eyalroz, 2022-01-07:
@JakeSays : Thanks for your input. In a sense, it is not entirely my decision, since I see myself more as a steward of this library than an owner. 

I wonder, though...

* What about the option of a single-header file as a build artifact, or perhaps available in _addition_ to the separate files?
* The codebase already has kind of a lot of `#ifdef`s for various features. Would this really make things much different?

@edubart : We now have 1 person for and 1 against, at the bottom line. So, expecting your answers to my questions. At any rate, I feel I need more people to chime in regarding this issue before we make any decisions.

edubart, 2022-01-08:
> We now have 1 person for and 1 against, at the bottom line. So, expecting your answers to my questions. At any rate, I feel I need more people to chime in regarding this issue before we make any decisions.

This was just a suggestion, I am not using this library at the moment because it's not single header, I am at the moment using the other ones mentioned in the issue, because they are, however I did consider this library. The code base I am working have a requirement to use single header libraries, to be able to bundle bundle them in a big C file, instead of touching a build system, but that is just my case. Do what you think it is the best for this project. I will continue using the other ones because I can update them by downloading just one header file from github.

eyalroz, 2022-01-08:
>  The code base I am working have a requirement to use single header libraries, to be able to bundle bundle them in a big C file, 

So, you're saying you're working on a project where you can't link against .o files?

Also, why can't you use the two files and just include them via the C file like I suggested? That will get you the single C file you wanted? Or is it just the matter of convenience of downloading a single file?

data-man, 2022-01-08:
+1 from me. :)

eyalroz, 2022-01-08:
@data-man : Since this is somewhat of a contentious issue, can you please elaborate...

* How a single-header is important for your use of the library (or whether you just find it convenient, which is legitimate of course)
* Whether you believe it's good enough to have this generated during build, and/or perhaps made available for release versions (and if not, why not).

data-man, 2022-01-08:
@eyalroz 

> whether you just find it convenient

Yes. @edubart described the reasons very well.

> Whether you believe it's good enough to have this generated during build

No. `printf.c` should be rewritten.

eyalroz, 2022-01-08:
@data-man : What if you got a single header file, as one of the release resources - alternative to getting the entire archive with the two files, the CMakeLists.txt, etc?

data-man, 2022-01-08:
Maybe.
Some project uses it, e.g. https://github.com/zeux/pugixml/blob/9e382f98076e57581fcc61323728443374889646/src/pugixml.hpp#L1473

phillipjohnston, 2022-01-24:
While it is a convention in the C++ world to provide a number of libraries as a single header, I have never heard of such a thing in C land at all. 

For this point:

> The code base I am working have a requirement to use single header libraries, to be able to bundle bundle them in a big C file,

Unity builds generally solve this problem without requiring everything to be in a header file, because the .c files are just included in one big .c file anyway.

> Also, why can't you use the two files and just include them via the C file like I suggested?

I agree that there is no functional difference between including the printf.c file in a "unity" .c file and having the contents in a header.

> > Whether you believe it's good enough to have this generated during build
> No. printf.c should be rewritten.

I'm not clear on why printf.c should be rewritten vs having a generated header you can download, what is the difference?

eyalroz, 2022-03-04:
If more users ask for this, let's reopen. For now I don't find that this is very useful.

closed 2022-03-04 by eyalroz