eyalroz/printf #84: Compile option to process floating point values as float

jrahlf, 2022-01-20:
I have a project where I use single precision floats, but no doubles.
Problem is, that printf processes all floating point values as doubles, even if the caller is content with single precision formatting. On devices without double-FPU, this pulls in `__aeabi_dsub`, `__aeabi_dmul` and the likes, which are several kilobytes large.

My suggestion is to add a compile option so that floating point values are formatted with single precision.

The value would still be passed as a double to `printf`, which would then be converted to float.
Some of the required functionality seems to be already there, via the `#if DBL_MANT_DIG == 24` define.

eyalroz, 2022-01-20:
I don't quite understand what you mean by:

> The value would still be passed as a double to printf, which would then be converted to float.

If doubles are still to be used, what does this suggestion save you?

eyalroz, 2022-01-20:
At any rate, if someone were to create a PR generalizing the internal floating-point-value handling to not necessarily use doubles, but some `printf_floating_point_t`, chosen via preprocessor defines, I would consider it. A PR would need to include appropriate tests of course.

jrahlf, 2022-01-22:
> If doubles are still to be used, what does this suggestion save you?

doubles would only be used for passing floating point values to `printf()`, because the standard mandates this for varargs. Passing floats directly would be even better for microcontrollers (if float precision is enough)...

The screenshot shows the added functions for a CM0+ microcontroller, this will be similar for any ARM microcontroller without dp-FPU.

[image]

- If the processing inside `printf()` was done on float instead of double, the compiler would not have to add these functions (assuming the code does not use doubles somewhere else).
  - On devices without FPU, the compiler would then use the `__aeabi_f` counterparts, but these are
    a) smaller
    b) more likely to be already in use

I will try to come up with a PR.

jrahlf, 2022-01-22:
Changing `double` to `floating_point_t` and adapting the existing `DBL_MANT_DIG` was easy, but there is a problem: The code uses 64-bit integer math and converts between `floating_point_t` and `int_fast64_t`. This latter also pulls in the double routines, UGH (see attached picture).

For testing I changed all `int_fast64_t` to `int_fast32_t` and it worked - `__aeabi_d*` no more. This of course breaks the code if the input is larger than `2^32-1`.

Exponential formatting without `int64<>float` conversions should be possible, [ryu-f2s](https://github.com/ulfjack/ryu/blob/master/ryu/f2s.c) seems to provide just that.

[image]

eyalroz, 2022-01-22:
> This of course breaks the code if the input is larger than 2^32-1.

Perhaps you could split the uses of the `int_fast64_t` type: For actual integers, use it as-is; and for floating-point work, use `int_fast32_t` instead.

eyalroz, 2022-01-24:
@jrahlf : Note I am about to land the rather significant PR #92  onto the develop branch. Would you like me to wait for a couple of days with that, for you to submit your own PR? Or should I just go ahead and let you do some rebasing?

jrahlf, 2022-01-24:
Go ahead, changing the handling of the `double_components` without duplicating several code pieces requires a bit more work, so it will take some time.

I measured the clock cycles on an CM0+ without FPU (measured with timer running at CPU frequency), all compiled with -O3:

| variant | snprintf("%g", 123.456f);  | snprintf("%g", 123.456e6f); |
| --- | --- | --- |
| printf from gcc nano specs | 11900  | 9900 |
| printf w/ double and int64 | 17800 | 20600 |
| printf w/ float and int64 | 14500 | 16900 |
| printf w/ float and int32 | 7100 | 9100 |

Interesting results, I will repeat them for a CM4 with FPU. I expect that the last variant should get significantly faster then.

Also interesting:
- float to double takes 48 cpu cycles
- double to float takes 74 cpu cycles

eyalroz, 2022-01-25:
Did you enable LTO, when compiling and when linking? I've not done so in the `CMakeLists.txt` for [this reason](https://gitlab.kitware.com/cmake/cmake/-/issues/23136), but that might change the various numbers.

jrahlf, 2022-01-25:
Here are the cycle count values for a CM4 with FPU, same methodology as above. Compiled without LTO because gcc then strips debug symbols. I would expect that LTO improves `printf` due to `putchar_`, but not `snprintf`.

| variant | snprintf("%g", 123.456f);  | snprintf("%g", 123.456e6f); |
| --- | --- | --- |
| printf from gcc nano specs | 7200  | 5600 |
| printf w/ double and int64 | 9400 | 9400 |
| printf w/ float and int64 | 3500 | 3500 |
| printf w/ float and int32 | 1500 | 1500 |

- float to double: 25 cycles
- double to float: 29 cycles

One can see that you can get a lot faster than the highly optimized stdlib because the latter must work on doubles.

eyalroz, 2022-01-25:
These are interesting stats! But actually, what are the "gcc nano specs"?

At any rate, I've merged my significant change (and the next version number will be 6.0.0), so you're welcome to resume to resume the PR work. Your argument is compelling :-)

jrahlf, 2022-01-25:
"gcc nano specs" refers to linking the newlib-nano standard library, which is the de facto used standard library for small microcontrollers (you have to explicitly enable floating point support for printf). 

It might be that newlib (no nano) has a faster printf thatn newlib-nano..

eyalroz, 2022-01-29:
Any news? We have quite a bunch of changes on the develop branch, so that a new release should occur soon. Have you made progress on the single-precision floating point processing?

eyalroz, 2022-01-31:
@jrahlf : ping.

jrahlf, 2022-01-31:
I have not forgotten about this, but only made little further progress. If you don't want to keep this open we can close and re-open when I come up with a PR.

jrahlf, 2022-02-06:
If I understand it correctly, the current version does not even need the 64-bit processing, as long as `PRINTF_MAX_INTEGRAL_DIGITS_FOR_DECIMAL <=9` holds true. Adding a conditional typedef there is easy.

```c
#if PRINTF_MAX_INTEGRAL_DIGITS_FOR_DECIMAL <= 9
typedef int_fast32_t fp_component;
#else
typedef int_fast64_t fp_component;
#endif
```

I pushed the first version to https://github.com/jrahlf/printf/commit/b7e1f52606c05bf945fde1749de28ac7171fbc8c . Tests for float processing are not added yet. Basically all `double_x` get replaced by `floating_point_x` or `fp_t`. 

I just wasted 30min trying to rebase this sh#t onto develop. Now it won't compile, I am done for today.

eyalroz, 2022-02-06:
@jrahlf : Sorry about the difficulty in rebasing. Like I'd said earlier, there have been some significant changes - not functionality-wise, but code-arrangement wise - over the past few weeks which make it difficult. I'll see what you've done and should probably be able to adapt it.

However - I believe you're misinterpreting what `PRINTF_MAX_INTEGRAL_DIGITS_FOR_DECIMAL` means. For example, if you print with `%e`, that definition is not used at all - and you would not expect the accuracy to drop because someone set `PRINTF_MAX_INTEGRAL_DIGITS_FOR_DECIMAL`. So the choice of fp_component cannot be determined by it.

eyalroz, 2022-02-06:
... Yes, so, it's what I expected you would write. Indeed, there's not that much to change beyond the typedef replacing double. Hope to get around to this within the next couple of days.

jrahlf, 2022-02-07:
> However - I believe you're misinterpreting what `PRINTF_MAX_INTEGRAL_DIGITS_FOR_DECIMAL` means. For example, if you print with `%e`, that definition is not used at all - and you would not expect the [...]

The integral value of formerly `double_components` can only exceed the range of int32 if the floating value to be formatted is greater than int32 and if fixed point formatting is selected. However, fixed point formatting is aborted if the value is greater than 1e9 (default threshold). I assume that is why the default threshold is 1e9?

But I see now, the fractional part is a bit more complicated.
For fixed point formatting, 32bit suffices if the decimal digits is less than 9 digits, i.e. up to ´%8.f`.
For exponential formatting, 32bit suffices if the precision is less than 9, i.e. true for float but wrong for double.

eyalroz, 2022-02-07:
Please have a look at #135 and possibly try out the branch.

eyalroz, 2022-02-21:
So, this will be fixed in the upcoming v6.0.0.

eyalroz, 2022-06-03:
So, I've neglected to resolve this... were you ok with the brach with the fix integrated? If you were, I'll merge it into the develop branch.

eyalroz, 2022-06-30:
@jrahlf : ping.

closed 2022-07-26 by eyalroz

eyalroz, 2023-01-24:
Well... the merging _might_ have fixed the internal float handling, but - some test cases are broken now, so we need to fix that.

commit 83f58118 (2022-02-07): Fixes #84: Choice of `float`/`double`

* Can now choose between using `double` and `float` internally, for floating-point work, using the `PRINTF_USE_DOUBLE_INTERNALLY` definition and corresponding CMake option.
* Adjusted test suite to support the different choice of floating-point type:
	* Tests relevant both to `float` and `double`, but with different precisions, are adjusted with an `#if #else #endif`
	* Tests and test cases relevant only for `double` precision are not compiled at all when `float` is used.
* Lots of new explicit conversions :-(
commit 61921be6 (2023-01-21): Fixes #84: Choice of `float`/`double`

* Can now choose between using `double` and `float` internally, for floating-point work, using the `PRINTF_USE_DOUBLE_INTERNALLY` definition and corresponding CMake option.
* Adjusted test suite to support the different choice of floating-point type:
	* Tests relevant both to `float` and `double`, but with different precisions, are adjusted with an `#if #else #endif`
	* Tests and test cases relevant only for `double` precision are not compiled at all when `float` is used.
* Lots of new explicit conversions :-(
PR #116 (2023-01-21): Fixes #84: Choice of `float`/`double`
Fixes #84.
commit a2dd4587 (2023-01-21): Fixes #84: Choice of `float`/`double`

* Can now choose between using `double` and `float` internally, for floating-point work, using the `PRINTF_USE_DOUBLE_INTERNALLY` definition and corresponding CMake option.
* Adjusted test suite to support the different choice of floating-point type:
	* Tests relevant both to `float` and `double`, but with different precisions, are adjusted with an `#if #else #endif`
	* Tests and test cases relevant only for `double` precision are not compiled at all when `float` is used.
* Lots of new explicit conversions :-(