eyalroz/printf #85: Test suite won't build with standard function aliasing on, due to catch

eyalroz, 2022-01-21:
The catch unit-testing framework includes  the C standard library's printf-family functions; consequently, it clashes with our library when we alias the standard printf-family function names. In the past this was circumvented by including `printf.c` within a namespace, but now that `printf.c` is more language-aware, that's not possible; plus, it wasn't a good idea, since it was testing something others than what library users will actually be doing.

Anyway, now we have this clash. This means either we need to not run the test suite with the aliased names, or ... do something else?

phillipjohnston, 2022-01-25:
Huh... I'm not using your build system (libc is using Meson), but I do have aliasing enabled and the tests build for me.

phillipjohnston, 2022-01-25:
I was wrong, looks like that compiler arg isn't applied to the test suite! Sorry for the distraction.

commit e014090d (2022-01-21): Fixes #85: Only compiling our test suite when standard library function name aliasing is disabled. For testing aliased functionality, we have a separate test.
commit dd06282d (2022-01-21): Fixes #85: Only compiling our test suite when standard library function name aliasing is disabled. For testing aliased functionality, we have a separate test.
commit b9b4bf15 (2022-01-22): Fixes #85: Only compiling our test suite when standard library function name aliasing is disabled. For testing aliased functionality, we have a separate test.
commit abcd93e9 (2022-01-26): Fixes #85: Only compiling our test suite when standard library function name aliasing is disabled. For testing aliased functionality, we have a separate test.
closed 2022-02-28 by eyalroz