eyalroz/printf #86: I have a problem there

bingheng-wlm, 2022-01-21:
printf.c:(.text+0x790): undefined reference to `__stack_chk_fail'
ld: printf.o: in function `print_decimal_number':
printf.c:(.text+0x12e6): undefined reference to `__stack_chk_fail'
ld: printf.o: in function `print_exponential_number':
printf.c:(.text+0x1a33): undefined reference to `__stack_chk_fail'
ld: printf.o: in function `print_floating_point':
printf.c:(.text+0x1d65): undefined reference to `__stack_chk_fail'
ld: printf.o:printf.c:(.text+0x33ce): more undefined references to `__stack_chk_fail' follow

just this.
So, ......
why?

eyalroz, 2022-01-21:
This is likely a problem with your compiler or linker.

How are you building the library? How are you linking your executable?

phillipjohnston, 2022-01-24:
If your standard library does not have a stack protector implementation, this will happen.

You can turn this off with -fno-stack-protector compiler argument, or you can supply an implementation for the stack protection functions: https://embeddedartistry.com/blog/2020/05/18/implementing-stack-smashing-protection-for-microcontrollers-and-embedded-artistrys-libc/

eyalroz, 2022-01-25:
@phillipjohnston : Can you explain what's going on? Is my library doing something problematic? Regardless, given what you've said, perhaps the `CMakeLists.txt` should detect this "lack of stack protector", when one is needed.

phillipjohnston, 2022-01-25:
@eyalroz if the software is built with -fstack-protector or similar flags (which may or may not be done by default), when certain conditions are detected the compiler will insert code to check whether or not a function is smashing the stack. Some implementations of the standard library don't come with this function (such as arm-gcc), so it will cause a linker error if it is not implemented but the compiler emitted those functions. My notes were meant for @bingheng-wen.

I don't really think it's a responsibility of your library to handle this, but if more people report it you could add a convenience option to disable stack smashing protection. (-fno-stack-protector for gcc/clang)

eyalroz, 2022-01-25:
Do you mean having that as a PRIVATE compilation flag for the `.a`/`.so` file, or as an INTERFACE compilation flag for code  using the library?

phillipjohnston, 2022-01-25:
Again, I think that it is proper for it to be handled outside of your library, but if you were to add it to your library it would be better to add it as a PRIVATE compilation flag.

eyalroz, 2022-01-26:
Agreeing with @phillipjohnston : It's not this library's fault or responsibility.

closed 2022-01-26 by eyalroz