eyalroz/printf #87: Silent narrowing conversion(s) from size_t to printf_size_t may bite us

eyalroz, 2022-01-22:
printf()-family functions have the questionable design choice of returning `int` as the number of characters printed, despite buffer lengths possibly being larger than `INT_MAX`. 

In our implementation, this is compounded by us using an int-sized `printf_size_t` type for offsets and lengths within the buffer, corresponding to the final return type - while taking in a `size_t` for functions such as `snprintf()`. At the moment, we simply cast from the latter into the former type at various locations, never checking for overflow. This could theoretically result in us getting lengths of 0, or other short lengths, which will mess up our behavior.

We need to cast as early as possible, and discard the `size_t`  inputs.

phillipjohnston, 2022-01-24:
There are still some implicit casts on the development branch - I will create a PR for those.

eyalroz, 2022-01-25:
So, your PR #97 had one such cast - have you noticed others?

Also, in my defense, that cast was not that implicit - it was in an assignment; and it's where I want the cast to happen. But the PR is merged.

phillipjohnston, 2022-01-25:
That was the only one, and I just used the warning term, not asking you to defend yourself :)

commit 69e6c94e (2022-01-22): Fixes #87,#91: Now explicitly converting from `size_t` to `printf_size_t` as soon as possible, taking care not to exceed the maximum representable value of `printf_size_t`. Also, no longer using `-1` magic numbers for the "maximum possible size/length".
commit 2575476a (2022-01-22): Fixes #87,#91: Now explicitly converting from `size_t` to `printf_size_t` as soon as possible, taking care not to exceed the maximum representable value of `printf_size_t`. From that point onwards (or rather inwards) - only `printf_size_t` is used for lengths and positions/indices. Also, no longer using `-1` magic numbers for the "maximum possible size/length".
commit 93c465cb (2022-01-24): Fixes #87,#91: Now explicitly converting from `size_t` to `printf_size_t` as soon as possible, taking care not to exceed the maximum representable value of `printf_size_t`. From that point onwards (or rather inwards) - only `printf_size_t` is used for lengths and positions/indices. Also, no longer using `-1` magic numbers for the "maximum possible size/length".
commit f4479dec (2022-01-26): Fixes #87,#91: Now explicitly converting from `size_t` to `printf_size_t` as soon as possible, taking care not to exceed the maximum representable value of `printf_size_t`. From that point onwards (or rather inwards) - only `printf_size_t` is used for lengths and positions/indices. Also, no longer using `-1` magic numbers for the "maximum possible size/length".
closed 2022-01-25 by eyalroz