eyalroz/printf #93: Missing tests for floating point with integral > 2^31 -1

jrahlf, 2022-01-23:
While working on #84, I noticed that the test coverage for large floating point values seems insufficient.
When changing the integral field type to `int32_t`, the test suite still passes. This makes me believe that there is no test case for floating point values with the integral part > `2^31 -1`.
https://github.com/eyalroz/printf/blob/36ad0167c1378907c07611b6b3d13ab3b07a78da/src/printf/printf.c#L423

eyalroz, 2022-01-23:
I'll have a look.

eyalroz, 2022-01-23:
@jrahlf : How's this?

jrahlf, 2022-01-25:
Now fails for 32-bit integral type as it should.

eyalroz, 2022-01-25:
@jrahlf : Hmm... should it, though? Or - do you mean, on the version you're working on, which uses 32-bit-integers only, internally?

jrahlf, 2022-01-25:
What I meant: if you change the type from `uint_fast64_t` to `uint32_t`, the test suite fails.

eyalroz, 2022-02-07:
I think we're ok now.

closed 2022-02-07 by eyalroz