eyalroz/printf #100: Float Equality Comparisons

phillipjohnston, 2022-01-24:
There are a number of -Wfloat-equal warnings that show up in the library. My typical 

```
../printf/src/printf/printf.c:542:26: warning: comparing floating point with == or != is unsafe [-Wfloat-equal]
  542 |     if (scaled_remainder == rounding_threshold) {
      |
```

I can prepare a PR that uses epsilon comparisons, but first I wanted to check whether or not you think that there is a reason these direct float comparisons are actually valid for some reason.

eyalroz, 2022-01-25:
> whether or not you think that there is a reason these direct float comparisons are actually valid for some reason.

They are all valid, and safe, in the context in which they are performed. If you believe some specific comparison is problematic, please reopen and say which, and why.

closed 2022-01-25 by eyalroz

phillipjohnston, 2022-01-25:
Cool, if they are all valid, I will prepare a change for the top of the source file that will turn off these warnings for GCC and Clang.

eyalroz, 2022-01-25:
Do you mean with #push and #pop directives? or via the CMakeLists.txt ?