eyalroz/printf #95: develop branch: limits.h is required for INT_MAX

phillipjohnston, 2022-01-24:
The develop branch fails to build for me at this moment (within my libc build) due to INT_MAX being referenced but not defined. I notice that limits.h is wrapped in an #ifdef, but it is needed unconditionally for INT_MAX.

```
#ifdef PRINTF_SUPPORT_MSVC_STYLE_INTEGER_SPECIFIERS
#include <limits.h>
#endif
```

phillipjohnston, 2022-01-24:
I reviewed everything and think you should just include those headers anyway - PR opened.

eyalroz, 2022-01-25:
Why do you believe I need those headers anyway? INT_MAX is only used when MSVC-style specifiers are supported.

phillipjohnston, 2022-01-25:
Without `PRINTF_SUPPORT_MSVC_STYLE_INTEGER_SPECIFIERS`, the code fails to compile for me because it is referencing `INT_MAX`.

```
FAILED: printf_tests.p/printf_test_test_suite.cpp.o
ccache c++ -Iprintf_tests.p -I. -I.. -I../printf/test -fcolor-diagnostics -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Wpedantic -std=c++11 -O2 -isystem../printf/src -MD -MQ printf_tests.p/printf_test_test_suite.cpp.o -MF printf_tests.p/printf_test_test_suite.cpp.o.d -o printf_tests.p/printf_test_test_suite.cpp.o -c ../printf/test/test_suite.cpp
In file included from ../printf/test/test_suite.cpp:30:
../printf/src/printf/printf.c:1171:67: error: use of undeclared identifier 'INT_MAX'
          printf_size_t l = strnlen_s_(p, precision ? precision : PRINTF_MAX_POSSIBLE_BUFFER_SIZE);
                                                                  ^
../printf/src/printf/printf.c:250:41: note: expanded from macro 'PRINTF_MAX_POSSIBLE_BUFFER_SIZE'
#define PRINTF_MAX_POSSIBLE_BUFFER_SIZE INT_MAX
                                        ^
../printf/src/printf/printf.c:1250:43: error: use of undeclared identifier 'INT_MAX'
  return _vsnprintf(&out_putchar, buffer, PRINTF_MAX_POSSIBLE_BUFFER_SIZE, format, arg);
                                          ^
../printf/src/printf/printf.c:250:41: note: expanded from macro 'PRINTF_MAX_POSSIBLE_BUFFER_SIZE'
#define PRINTF_MAX_POSSIBLE_BUFFER_SIZE INT_MAX
                                        ^
../printf/src/printf/printf.c:1255:36: error: use of undeclared identifier 'INT_MAX'
  return _vsnprintf(out_buffer, s, PRINTF_MAX_POSSIBLE_BUFFER_SIZE, format, arg);
                                   ^
../printf/src/printf/printf.c:250:41: note: expanded from macro 'PRINTF_MAX_POSSIBLE_BUFFER_SIZE'
#define PRINTF_MAX_POSSIBLE_BUFFER_SIZE INT_MAX
                                        ^
../printf/src/printf/printf.c:1260:43: error: use of undeclared identifier 'INT_MAX'
  printf_size_t usable_buffer_size = (n > PRINTF_MAX_POSSIBLE_BUFFER_SIZE) ? PRINTF_MAX_POSSIBLE_BUFFER_SIZE : n;
                                          ^
../printf/src/printf/printf.c:250:41: note: expanded from macro 'PRINTF_MAX_POSSIBLE_BUFFER_SIZE'
#define PRINTF_MAX_POSSIBLE_BUFFER_SIZE INT_MAX
                                        ^
../printf/src/printf/printf.c:1260:78: error: use of undeclared identifier 'INT_MAX'
  printf_size_t usable_buffer_size = (n > PRINTF_MAX_POSSIBLE_BUFFER_SIZE) ? PRINTF_MAX_POSSIBLE_BUFFER_SIZE : n;
                                                                             ^
../printf/src/printf/printf.c:250:41: note: expanded from macro 'PRINTF_MAX_POSSIBLE_BUFFER_SIZE'
#define PRINTF_MAX_POSSIBLE_BUFFER_SIZE INT_MAX
                                        ^
../printf/src/printf/printf.c:1267:76: error: use of undeclared identifier 'INT_MAX'
  return _vsnprintf(out_wrapped_function, (char*)(uintptr_t)&out_fct_wrap, PRINTF_MAX_POSSIBLE_BUFFER_SIZE, format, arg);
                                                                           ^
../printf/src/printf/printf.c:250:41: note: expanded from macro 'PRINTF_MAX_POSSIBLE_BUFFER_SIZE'
#define PRINTF_MAX_POSSIBLE_BUFFER_SIZE INT_MAX
                                        ^
```

phillipjohnston, 2022-01-25:
Around line 244 in my fork's current state:

```
// The printf()-family functions return an `int`; it is therefore
// unnecessary/inappropriate to use size_t - often larger than int
// in practice - for non-negative related values, such as widths,
// precisions, offsets into buffers used for printing and the sizes
// of these buffers. instead, we use:
typedef unsigned int printf_size_t;
#define PRINTF_MAX_POSSIBLE_BUFFER_SIZE INT_MAX
  // If we were to nitpick, this would actually be INT_MAX + 1,
  // since INT_MAX is the maximum return value, which excludes the
  // trailing '\0'.
```

commit ead64836 (2022-01-25): Fixes #95: Now always including `limits.h` (as we use `INT_MAX` as a default upper bound on buffer sizes, due to the `int` return type of printf()-family functions). Thanks goes to Phillip Johnston <phillip@embeddedartistry.com>.
commit 046b74c5 (2022-01-26): Fixes #95: Now always including `limits.h` (as we use `INT_MAX` as a default upper bound on buffer sizes, due to the `int` return type of printf()-family functions). Thanks goes to Phillip Johnston <phillip@embeddedartistry.com>.
commit 79da9709 (2022-01-31): Fixes #95: Now always including `limits.h` (as we use `INT_MAX` as a default upper bound on buffer sizes, due to the `int` return type of printf()-family functions). Thanks goes to Phillip Johnston <phillip@embeddedartistry.com>.
commit a7ace464 (2022-01-31): Fixes #95: Now always including `limits.h` (as we use `INT_MAX` as a default upper bound on buffer sizes, due to the `int` return type of printf()-family functions). Thanks goes to Phillip Johnston <phillip@embeddedartistry.com>.
closed 2022-02-28 by eyalroz