eyalroz/printf #105: GCC Conversion Warnings

phillipjohnston, 2022-01-25:
GCC with -Wconversion highlights conversions that may change values.

```
[243/533] Compiling C object src/libc.a.p/.._printf_src_printf_printf.c.o
../printf/src/printf/printf.c: In function 'get_sign':
../printf/src/printf/printf.c:275:30: warning: conversion from 'double_uint_t' {aka 'long long unsigned int'} to 'int' may change value [-Wconversion]
  275 |   return get_bit_access(x).U >> (DOUBLE_SIZE_IN_BITS - 1);
      |          ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
../printf/src/printf/printf.c: In function 'get_components':
../printf/src/printf/printf.c:528:34: warning: conversion from 'int_fast64_t' {aka 'long long int'} to 'double' may change value [-Wconversion]
  528 |   double remainder = (abs_number - number_.integral) * powers_of_10[precision];
      |                                  ^
../printf/src/printf/printf.c:618:22: warning: conversion from 'int_fast64_t' {aka 'long long int'} to 'double' may change value [-Wconversion]
  618 |     scaled_remainder -= components.fractional;
      |                      ^~
```

I can resolve these in a PR with the proper casts, but I wanted you to take a look first in case you think some of them should be resolved through other means.

phillipjohnston, 2022-01-25:
(This is on develop)

eyalroz, 2022-01-25:
Fixed this, and added `-Wconversion`  myself. When a warning can just be avoided, I have no trouble obliging.

commit 12f14336 (2022-01-25): Fixes #105: Making all potentially-value-altering conversions explicit, and some additional minor changes:

* Now compiling with the `-Wconversion` warning flag when using clang or gcc.
* In all cases noticed by gcc in which an implicit conversion was made which could change the converted value - have changed the conversion to be explicit.
* Renamed: `get_sign()` -> `get_sign_bit()` (so as to clarify the return value is either 0 or 1, not -1 or 1).
* Added explanatory comment for the `double_components` structure.
commit 55d97e9a (2022-01-26): Fixes #105: Making all potentially-value-altering conversions explicit, and some additional minor changes:

* Now compiling with the `-Wconversion` warning flag when using clang or gcc.
* In all cases noticed by gcc in which an implicit conversion was made which could change the converted value - have changed the conversion to be explicit.
* Renamed: `get_sign()` -> `get_sign_bit()` (so as to clarify the return value is either 0 or 1, not -1 or 1).
* Added explanatory comment for the `double_components` structure.
commit 1310d53a (2022-01-31): Fixes #105: Making all potentially-value-altering conversions explicit, and some additional minor changes:

* Now compiling with the `-Wconversion` warning flag when using clang or gcc.
* In all cases noticed by gcc in which an implicit conversion was made which could change the converted value - have changed the conversion to be explicit.
* Renamed: `get_sign()` -> `get_sign_bit()` (so as to clarify the return value is either 0 or 1, not -1 or 1).
* Added explanatory comment for the `double_components` structure.
commit 2f31dfb9 (2022-01-31): Fixes #105: Making all potentially-value-altering conversions explicit, and some additional minor changes:

* Now compiling with the `-Wconversion` warning flag when using clang or gcc.
* In all cases noticed by gcc in which an implicit conversion was made which could change the converted value - have changed the conversion to be explicit.
* Renamed: `get_sign()` -> `get_sign_bit()` (so as to clarify the return value is either 0 or 1, not -1 or 1).
* Added explanatory comment for the `double_components` structure.
closed 2022-02-28 by eyalroz