eyalroz/printf #117: Undefined behavior using va_arg in nested function

dsludwig, 2022-02-11:
As of https://github.com/eyalroz/printf/commit/64042838abfd0b40bfb6e23005f80598aa114e37, the call to `va_arg` for integers is no longer in `_vsnprintf`. This creates undefined behavior, as described in the C99 standard:

> The object ap may be passed as an argument to another function; if that function invokes the va_arg macro with parameter ap, the value of ap in the calling function is indeterminate and shall be passed to the va_end macro prior to any further reference to ap

http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1124.pdf
https://stackoverflow.com/a/3369762

This resulted in the same variable being printed for each specifier in my environment (GNU Arm Embedded Toolchain 10.3-2021.10).

```
// 1:1 2:1 3:1
printf("1:%i 2:%i 3:%i", 1, 2, 3);
```

I didn't experience this issue when compiling for x86-64, I suspect this may be affected by the inlining decision of the compiler (especially if `-ffunction-sections` is used).

Affects 5.3.0 & 6.0.0.

mickjc750, 2022-02-11:
I had to face this problem in my prnf(). 
 The implementation of va_arg differs between compilers. On GCC-AVR I experienced the same symptom as you, each call accessed the first variable, and on x86-64 it worked. I thought passing a pointer to va_arg would fix it, which it does on AVR, but this actually breaks it on x86-64. This is because the x86-64 implementation uses an array type which gets broken by pointer decay. To make my code portable I had to scan the entire va_arg list within a single function.

eyalroz, 2022-02-11:
@dsludwig : Thank you for reporting this. I suppose there is nothing I can do other than inline the `process_integer_specifier()` function back into `_vsprintf()`?

eyalroz, 2022-02-11:
Fixed this - I think - by altering the offending commit. Please re-download 5.3.0 and/or 6.0.0 to verify.

closed 2022-02-11 by eyalroz