eyalroz/printf #120: remove “0x” prefix

thewon86, 2022-02-22:
int a = 0x5a5a5a5a;
printf("%x", a);
> 0x5a5a5a5a ?

how could i remove the prefix when using x X p format?

thx

eyalroz, 2022-02-22:
That's not the output you get. `printf("%x",a)` yields `5a5a5a5a`, while `printf("%#x",a)` yields `0x5a5a5a5a`.

closed 2022-02-22 by eyalroz

thewon86, 2022-02-22:
> That's not the output you get. `printf("%x",a)` yields `5a5a5a5a`, while `printf("%#x",a)` yields `0x5a5a5a5a`.

i found the code
https://github.com/eyalroz/printf/blob/3f6d9725ab50906c107028b5c75b016adad2ce81/src/printf/printf.c#L360-#L368 
and
https://github.com/eyalroz/printf/blob/3f6d9725ab50906c107028b5c75b016adad2ce81/src/printf/printf.c#L1037

comment L360-L368, modify L1037 `width = sizeof(void*) * 2U;`, now solved it.

mickjc750, 2022-02-23:
@thewon86 
If you still believe printf("%x",a) yields 0x5a5a5a5a , I would suggest making an obvious change like:

printf("--->  %x  <---",a);

And I think you'll find that this isn't the line being executed. You most likely have a "%#x" somewhere else in your source.