eyalroz/printf #123: Uninitialised variable in print_exponential_number

mattbucknall, 2022-03-01:
Hi - I get the following warning when compiling printf.c (master [8b831c1]):

printf.c:899:47: warning: 'abs_exp10_covered_by_powers_table' may be used uninitialized in this function [-Wmaybe-uninitialized]

I've not done any in-depth analysis of how print_exponential_number works, so I'm not sure if this really matters. Still, it would be nice if abs_exp10_covered_by_powers_table could be initialised with a default value to get rid of this warning.

eyalroz, 2022-03-01:
This is a compiler "bug". Look at the logic:
```
  // ... snip ...
  if (abs_number == 0.0) {
    floored_exp10 = 0;
  }
  else  {
    // ... snip ...
    normalization.raw_factor = // ... snip ...
  }
  // ... snip ...
  normalization.multiply = (floored_exp10 < 0 && abs_exp10_covered_by_powers_table);
  // ... snip ...
```
You see? It can't be used uninitialized.

What compiler are you using?

closed 2022-03-01 by eyalroz

203Null, 2022-04-08:
I have the same issue with using this library with ESP-IDF

[image]

eyalroz, 2022-04-08:
> I have the same issue with using this library with ESP-IDF

I don't know what ESP-IDF means. Please file a bug against either the IDE or the compiler it uses to generate warnings. There is no issue here.

203Null, 2022-04-08:
> I don't know what ESP-IDF means. Please file a bug against either the IDE or the compiler it uses to generate warnings.

It will not compile with -Wmaybe-uninitialized unless I default abs_exp10_covered_by_powers_table to false.

203Null, 2022-04-08:
[image]

eyalroz, 2022-04-08:
First of all, you're treating warnings as errors. That is your prerogative, but the library's build mechanism does not do that on its own.

Secondly, and again - a warning here is a bug in your compiler. There is no sense in complaining about it here. The code is correct. Please file a bug against the compiler.

203Null, 2022-04-08:
> First of all, you're treating warnings as errors. That is your prerogative, but the library's build mechanism does not do that on its own.
>
> Secondly, and again - a warning here is a bug in your [...]

The only place that abs_exp10_covered_by_powers_table is defined is at https://github.com/eyalroz/printf/blob/19a487cd210d7de0914809fdf5d8bb45af1656c7/src/printf/printf.c#L873
which is under an if-else statement. I don't see how this is a compiler error.

203Null, 2022-04-08:
I see, the code won't check for abs_exp10_covered_by_powers_table if floored_exp10  == 0

eyalroz, 2022-04-08:
@203Null : Exactly.

203Null, 2022-04-08:
The compiler I used is Xtensa ESP32 gcc 8.4.0 (2021r2)
On GodBolt with the same compiler, it works fine https://godbolt.org/z/ofqnoWbzM

I will make an issue in their Github repo

203Null, 2022-11-03:
I saw you did some commits regarding this issue. 
https://github.com/eyalroz/printf/blob/f8ed5a9bd9fa8384430973465e94aa14c925872d/src/printf/printf.c#L631
Those codes are actually fine with my compiler (not sure about others)

These lines in print_exponential_number are actually the ones causing issues for me. Porting your changes over solves it.
https://github.com/eyalroz/printf/blob/f8ed5a9bd9fa8384430973465e94aa14c925872d/src/printf/printf.c#L908

203Null, 2022-11-06:
> @203Null : Please open a separate about your issue; explain exactly on which system and with which compiler in what version you get an error, and copy the relevant compiler output.

It's the exact same issue. I posted my log earlier https://github.com/eyalroz/printf/issues/123?notification_referrer_id=NT_kwDOATAnpbMzMTk2MDU5NTAzOjE5OTMzMDkz#issuecomment-1093291418

Just want to point out the ` #pragma GCC diagnostic ignored "-Wmaybe-uninitialized" ` was added to the wrong place (I assume you added them to fix this issue judging by the commit name.)