eyalroz/printf #125: Platform incompatible

mateoconlechuga, 2022-03-28:
Hello, currently there is a note in the readme that says:

> The` implementation currently assumes each of intmax_t, signed size_t, and ptrdiff_t has the same size as long int or as long long int. If this is not the case for your platform, please open an issue.

Currently our platform has `intmax_t`, signed `size_t`, and `ptrdiff_t` with the size of `int` only, which is 24-bits. Is there any hope of a workaround?

eyalroz, 2022-03-28:
I'm not sure I understand. `intmax_t` is supposed to be the largest integer type. Are you saying that on your platform, it is not the largest type? And that `long int` is larger than `intmax_t`? That sounds like an incompatibility with the C standard.

mateoconlechuga, 2022-03-28:
Sorry I was thinking of INT_MAX not the type - `intmax_t` is `long long`. `size_t` and `ptrdiff_t` are still computed as `int`.

eyalroz, 2022-03-28:
@mateoconlechuga : Then - the library should probably work fine on your system. Please try building it and running the test suite to check.

closed 2022-08-19 by eyalroz