eyalroz/printf #126: printf.h: definition of ATTR_PRINTF is invalid without __GNUC__

smcpeak, 2022-04-21:
As of commit 87878b20a5, `printf.h` when preprocessed without `__GNUC__` has this macro definition:

```
# define ATTR_PRINTF((one_based_format_index), (first_arg))
```

This is invalid since the macro parameters have extra parentheses.  `cpp -undef` (which does not set `__GNUC__`) complains:

```
$ cpp -undef printf.h >/dev/null
printf.h:56:22: error: expected parameter name, found "("
   56 | # define ATTR_PRINTF((one_based_format_index), (first_arg))
      |                      ^
Exit 1
$ cpp --version
cpp (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
```

In the original commit, that line should not have been changed.

Ideally, there should also be an automated test that exercises this printf implementation without `__GNUC__`, as that would have caught this error.

eyalroz, 2022-04-22:
First, thank you for reporting this bug.

Now, would you like to submit a PR? Indeed, I currently only test on GNUC-ish compilers. Perhaps I should work on creating an MSVC workflow.

smcpeak, 2022-04-22:
I hesitate to submit a PR since (ideally) it should include a test and I'm not sure what would fit best in this project.  (And, as I'm working on other things, I'm not able to properly explore that currently.)

However, as for how to test, one suggestion is to preprocess with `cpp -undef` to get a "compiler-agnostic" version and then feed that to GCC.  That's effectively what I was doing when I ran into this.

eyalroz, 2022-04-22:
@smcpeak : Well, I don't know about `cpp -undef`. It` will break the library in a dozen ways. I think it undefines more than you might expect.

commit 88eba51b (2022-04-22): Fixes #126: Corrects a bug introduced in commit 87878b2 for non-GNUC compilers, where macro arguments were mistakenly wrapped in parentheses.
closed 2022-10-17 by eyalroz