eyalroz/printf #138: Weird integers printing behavior on bare metal embedded system

ShellCode33, 2022-07-05:
Hey, thanks for this printf implementation which replaced the old one we were using which sucked really hard.

I'm not sure you'll be able to help because there's very little I can provide to help you understand what's going on. And this is probably an issue on our side (memory corruption?) which is not related at all to your printf implementation. But still, I wanted to ask, maybe you can give me valuable insights that will help me understand what's going on.

First, a bit of context.

We're doing hardware simulation of a [Pulpino CPU](https://github.com/pulp-platform/pulpino/) which is a RISCV architecture.

Our code is being compiled using a pretty old version of gcc :
```
$ riscv32-unknown-elf-gcc --version
riscv32-unknown-elf-gcc (GCC) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
We're not using CMake so we copy pasted your printf source code into ours and configured it as mentioned in your README.
Here's our printf configuration :

```
// SUPPORT CONFIG
#define PRINTF_SUPPORT_DECIMAL_SPECIFIERS            0
#define PRINTF_SUPPORT_EXPONENTIAL_SPECIFIERS        0
#define PRINTF_SUPPORT_WRITEBACK_SPECIFIER           0
#define PRINTF_SUPPORT_MSVC_STYLE_INTEGER_SPECIFIERS 0
#define PRINTF_SUPPORT_LONG_LONG                     1
#define PRINTF_ALIAS_STANDARD_FUNCTION_NAMES         1

// SPECIFIC CONFIG
#define PRINTF_INTEGER_BUFFER_SIZE              32
#define PRINTF_DECIMAL_BUFFER_SIZE              32
#define PRINTF_DEFAULT_FLOAT_PRECISION          6
#define PRINTF_MAX_INTEGRAL_DIGITS_FOR_DECIMAL  9
#define PRINTF_LOG10_TAYLOR_TERMS               4
```

Here's the code I've been using to "test" the printf implementation on our RISCV CPU :

```c
uint8_t u_eight = UINT8_MAX;
uint16_t u_sixteen = UINT16_MAX;
uint32_t u_thirty_two = UINT32_MAX;
uint64_t u_sixty_four = UINT64_MAX;

int8_t eight = INT8_MIN;
int16_t sixteen = INT16_MIN;
int32_t thirty_two = INT32_MIN;
int64_t sixty_four = INT64_MIN;

int neg = -42;
unsigned int pos = 42;

printf("\n");

printf("uint8_t = %" PRIu8 "\n", u_eight);
printf("uint16_t = %" PRIu16 "\n", u_sixteen);
printf("uint32_t = %" PRIu32 "\n", u_thirty_two);
printf("uint64_t = %" PRIu64 "\n", u_sixty_four);
printf("uint8_t = 0x%" PRIX8 "\n", u_eight);
printf("uint16_t = 0x%" PRIX16 "\n", u_sixteen);
printf("uint32_t = 0x%" PRIX32 "\n", u_thirty_two);
printf("uint64_t = 0x%" PRIX64 "\n", u_sixty_four);

printf("\n");

printf("int8_t = %" PRId8 "\n", eight);
printf("int16_t = %" PRId16 "\n", sixteen);
printf("int32_t = %" PRId32 "\n", thirty_two);
printf("int64_t = %" PRId64 "\n", sixty_four);
printf("int8_t = 0x%" PRIX8 "\n", eight);
printf("int16_t = 0x%" PRIX16 "\n", sixteen);
printf("int32_t = 0x%" PRIX32 "\n", thirty_two);
printf("int64_t = 0x%" PRIX64 "\n", sixty_four);

printf("\n");

printf("int = %d\n", neg);
printf("unsigned int = %u\n", 42);

printf("\n");
```

Here's the result :

```
# RX string: uint8_t = rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrt                                                                                                                                                                                                                      
# RX string: uint16_t = rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrt                                                                                                                                                                                                                     
# RX string: uint32_t = rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrt                                                                                                                                                                                                                     
# RX string: uint64_t = rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrt                                                                                                                                                                                                                     
# RX string: uint8_t = 0xdddddddddddddddddddddddddddddddf                                                                                                                                                                                                                    
# RX string: uint16_t = 0xdddddddddddddddddddddddddddddddf                                                                                                                                                                                                                   
# RX string: uint32_t = 0xdddddddddddddddddddddddddddddddf                                                                                                                                                                                                                   
# RX string: uint64_t = 0xdddddddddddddddddddddddddddddddf                                                                                                                                                                                                                   
# RX string:                                                                                                                                                                                                                                                                 
# RX string: int8_t = rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrõ                                                                                                                                                                                                                       
# RX string: int16_t = rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrru                                                                                                                                                                                                                      
# RX string: int32_t = rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrru                                                                                                                                                                                                                      
# RX string: int64_t = rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrru                                                                                                                                                                                                                      
# RX string: int8_t = 0xdddddddddddddddddddddddddddddddç                                                                                                                                                                                                                     
# RX string: int16_t = 0xdddddddddddddddddddddddddddddddg                                                                                                                                                                                                                    
# RX string: int32_t = 0xdddddddddddddddddddddddddddddddg                                                                                                                                                                                                                    
# RX string: int64_t = 0xdddddddddddddddddddddddddddddddg                                                                                                                                                                                                                    
# RX string:                                                                                                                                                                                                                                                                 
# RX string: int = rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrr                                                                                                                                                                                                                          
# RX string: unsigned int = rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrr 
```

Any clue what could be going on ? Any tips on how to debug this ? (our debug environment is very limited, we cannot attach a gdb debugger). I find it weird that letters are printed instead of digits. I would think of a memory corruption if the values were wrong, but letters instead of digits ? I don't get it...

Any help would be very appreciated ! Thanks :)

eyalroz, 2022-07-05:
Do you get this kind of behavior whenever you print _anything_, or is it just certain values that trigger it?

mickjc750, 2022-07-05:
The digits are produced by the do-while loop at lines 544-548 (in the function print_integer). This loop also counts the digits produced, and in your case it appears to always hit the limit of PRINTF_INTEGER_BUFFER_SIZE (your numeric outputs are all 32 characters).
 You could confirm this by changing PRINTF_INTEGER_BUFFER_SIZE and seeing that the your 'rrrrr' strings also change in length.
I would suggest trying to isolate this function in your own test, or even just the do-while loop.

eyalroz, 2022-07-06:
I second @mickjc750 's suggestion, but let me clarify a little: We have the `XXXprintf()` functions - those are wrappers, which parse the format strings and execute code to print values of specific types. In your case, it's integer printing that's problematic: [`print_integer()`](https://github.com/eyalroz/printf/blob/master/src/printf/printf.c#L524). 

To isolate it, you need to create an "output gadget" (see in printf.c), pass the value you want to print, and set the relevant flags: not negative, base 10, field width (you can set it the maximum possible, and also the flags, which you need to set carefully (see [here](https://github.com/eyalroz/printf/blob/master/src/printf/printf.c#L140)).
```
static void print_integer(output_gadget_t* output, printf_unsigned_value_t value, bool negative, numeric_base_t base, printf_size_t precision, printf_size_t width, printf_flags_t flags)
```
And of course you will need your code to be appended to `printf.c` 's context, to be able to use this function, as it is not exposed via `printf.h`.

ShellCode33, 2022-07-06:
Thanks for your comments, the investigation is still ongoing. Apparently the issue is somehow related to `PRINTF_SUPPORT_LONG_LONG`. When I disable it, everything works great, except I cannot print 64 bits integers, which is an issue.

I don't know if this is relevant to you but FYI the toolchain we are using defines `long` to be 4 bytes wide and `long long` to be 8.

```
$ echo | riscv32-unknown-elf-gcc -dM -E - | grep SIZEOF                                                                                              
#define __SIZEOF_INT__ 4
#define __SIZEOF_POINTER__ 4
#define __SIZEOF_LONG__ 4
#define __SIZEOF_LONG_DOUBLE__ 8
#define __SIZEOF_SIZE_T__ 4
#define __SIZEOF_WINT_T__ 4
#define __SIZEOF_PTRDIFF_T__ 4
#define __SIZEOF_FLOAT__ 4
#define __SIZEOF_SHORT__ 2
#define __SIZEOF_WCHAR_T__ 4
#define __SIZEOF_DOUBLE__ 8
#define __SIZEOF_LONG_LONG__ 8
```

eyalroz, 2022-07-06:
> I don't know if this is relevant to you but FYI the toolchain we are using defines long to be 4 bytes wide and long long to be 8.

That should not be an issue, but I'll double-check.

> Any kind of integer basically

Even if you print values like 0 or 1? What about negative values for signed integers?

ShellCode33, 2022-07-06:
So... I'm trying to debug the `do{}while` loop as you suggested, and I found something really really weird :

Here's my input test :
```c
uint32_t plop = 42;
printf("This is a test: %" PRIu32 "\n", plop);
```

Here's what I did in the do while loop :
```c
alt_print("value == 42 ? ");                                                      
alt_print(value == 42 ? "YES\n" : "NO\n");                                        
alt_print("base == 10 ? ");                                                       
alt_print(base == 10 ? "YES\n" : "NO\n");                                         
const char digit = (char)(value % base);                                          
alt_print("digit > 10 ? ");                                                       
alt_print(digit > 10 ? "YES\n" : "NO\n");    
```
(pretty ridiculous I know, but as I said, I have no way to debug this properly so....)

Here's what it gives me 
```
# RX string: value == 42 ? YES                                                                                                                                                                                                                                               
# RX string: base == 10 ? YES                                                                                                                                                                                                                                                
# RX string: digit > 10 ? YES
# RX string: value == 42 ? NO
# RX string: base == 10 ? YES                                                                                                                                                                                                                                                
# RX string: digit > 10 ? YES
[...]
# RX string: rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrt
```
42 % 10 > 10 ????

At first I thought maybe it's related to the cast not being `(unsigned char)` I know I had issues with that in the past (with MSVC I believe), but it didn't change anything.

When I disable `PRINTF_SUPPORT_LONG_LONG`, it looks fine and the digits are displayed properly :
```
# RX string: value == 42 ? YES                                                                                                                                                                                                                                               
# RX string: base == 10 ? YES                                                                                                                                                                                                                                                
# RX string: digit > 10 ? NO                                                                                                                                                                                                                                                 
# RX string: value == 42 ? NO                                                                                                                                                                                                                                                
# RX string: base == 10 ? YES                                                                                                                                                                                                                                                
# RX string: digit > 10 ? NO                                                                                                                                                                                                                                                 
# RX string: 42
```

ShellCode33, 2022-07-06:
> Even if you print values like 0 or 1? What about negative values for signed integers?

```c
uint32_t test0 = 0;
uint32_t test1 = 1;
int32_t test_neg = -42;

printf("test0: %" PRIu32 "\n", test0);
printf("test1: %" PRIu32 "\n", test1);
printf("test_neg: %" PRIi32 "\n", test_neg);
```

```
# RX string: test0: 0                                                                                                                                                                                                                                                        
# RX string: test1: rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrrv                                                                                                                                                                                                                         
# RX string: test_neg: rrrrrrrrrrrrrrrrrrrrrrrrrrrrrrr   
```

ShellCode33, 2022-07-06:
I'm unable to make any sense out of it... I created a `my_mod()` function to isolate the modulo operation.

Here's what the assembly looks like :
```asm
00006d3c <my_mod>:                                                              
     6d3c:    ff010113      addi    sp,sp,-16                                       
     6d40:    00112623      sw  ra,12(sp)                                     
     6d44:    d99fa0ef      jal 1adc <__umoddi3>                                    
     6d48:    00c12083      lw  ra,12(sp)                                           
     6d4c:    01010113      addi    sp,sp,16                                        
     6d50:    00008067      ret           
```

According to what I've read, `__umoddi3()` is responsible for 64 bits modulo operations, are you familiar with this function ? Could it be that the implementation we are using (from our toolchain's minimalist libc) is wrong ?

Also, I tried to perform the modulo operation manually as you suggested (`x_mod_y = x - (x / y)*y`) but gcc optimizes it to the exact same assembly as above.

Anyway, I think this confirms the issue comes from us. Maybe I will just disable 64 bits printf support. Thanks for your time, I don't want to waste any more of it. Feel free to close the issue.

eyalroz, 2022-07-06:
> but gcc optimizes it to the exact same assembly as above.

I think there should be a way to tell GCC to avoid optimizations locally.... ah, yes, [here it is](https://stackoverflow.com/a/2220565/1593077).

ShellCode33, 2022-07-07:
I tried disabling the optimization with no luck, `__umoddi3` was not called anymore but the `/` operator was still replaced with a call to `__udivdi3()`, the  result was still wrong, not 100% sure this bug is related to these functions though.

Disabling the constants merging using the flag `-fno-merge-all-constants` has fixed the issue. I'm blown away. In the end I don't know what went wrong but what I know for sure is that our toolchain sucks.

Again, thanks for your help.

eyalroz, 2022-07-07:
Well, if you want to continue narrowing this down to a minimally-failing program, feel free to reopen the bug.

closed 2022-07-07 by eyalroz

dbekatli, 2023-08-09:
@ShellCode33 I don't know whether this is still of interest to you but I also had a similar problem. The platform I'm using/developing is based on [Ibex](https://github.com/pulp-platform/ibex) and I'm using a self-built version of the latest [RISC-V GNU toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain) so perhaps our problems might be related. 

Trying to print uintX_t types either made my test program immediately cause an exception or yielded weird outputs which looked like binary numbers but had nothing to do with the values given to printf. The printf function worked fine without long long support and I could print uintX_t variables by casting them to `unsigned int`, however `-fno-merge-all-constants` didn't change anything in my case.

 The problem turned out to be my start file, which is supposed to initialize to global pointer and call main, wasn't actually working properly. The line initializing the gp register was turned into `mv gp, gp` due to linker relaxation and the problem went away after compiling the start file with the no_relaxation option. No idea why printf worked without long long support despite the obviously wrong gp value or whether this is the same issue you're having but I hope this helps.

ShellCode33, 2023-08-09:
I don't work for this company anymore, but good to know there's finally an explanation to this madness ;) Thanks for your feedback, maybe it will help other people ending up here

eyalroz, 2023-08-09:
@ShellCode33 Thank you for posting this comment. And it turns out that, rather than a bug, the library managed to work when it shouldn't have; so - it's an anti-bug :-P