eyalroz/printf #146: Issue with HARD aliasing the standard function names

BenFreudbergKolmostar, 2023-01-18:
It looks like there is some inconsistency introduced with the addition of `PRINTF_ALIAS_STANDARD_FUNCTION_NAMES_SOFT` which means that if you want to do HARD replacement, you must include both `PRINTF_ALIAS_STANDARD_FUNCTION_NAMES=1` and `PRINTF_ALIAS_STANDARD_FUNCTION_NAMES_HARD=1`. I fixed this on my local copy by getting rid of `PRINTF_ALIAS_STANDARD_FUNCTION_NAMES` altogether.

I deleted:
https://github.com/eyalroz/printf/blob/f8ed5a9bd9fa8384430973465e94aa14c925872d/src/printf/printf.h#L65-L67

and I changed:
https://github.com/eyalroz/printf/blob/f8ed5a9bd9fa8384430973465e94aa14c925872d/src/printf/printf.c#L57
to
```c
#if PRINTF_ALIAS_STANDARD_FUNCTION_NAMES_HARD
```

The documentation could also use an update to clarify this new feature.

Thanks!

eyalroz, 2023-01-21:
@BenFreudbergKolmostar : Please check whether my commit to the development branch resolves the issue to your satisfaction.

BenFreudbergKolmostar, 2023-01-23:
@eyalroz Yes, I think it would. But shouldn't printf.c:line57 be "HARD", not "SOFT"? I tried to leave this comment over there directly but I'm not sure if it worked.

Also, unrelated, but line 101/102 looks like a typo with two "INTERNALLY"s.

eyalroz, 2023-01-24:
@BenFreudbergKolmostar : Did see the comment. Please pull again (and note that previous commits are overridden now.

BenFreudbergKolmostar, 2023-01-24:
@eyalroz Looks great, thanks!

closed 2024-07-19 by eyalroz