eyalroz/printf #148: I want to add support for vector format

esrrhs, 2023-04-13:
Hi, after read the source code, I found that it can easy support format a variable array params by modify the ```vsnprintf_impl``` function. 
just wrap the ```va_list args``` to a input struct ```input_gadget_t```, like the ```output_gadget_t```. the  ```input_gadget_t``` then pop int/long/string one by one in the main loop code.

if that looks ok to you, I will make a pull request.

thank you

eyalroz, 2023-04-13:
I don't quite understand what you mean. What are "variable array parameters"? What missing feature do you want to implement?

closed 2023-04-13 by eyalroz

esrrhs, 2023-04-14:
sorry, my english make you confuse.

the pull request change is certainly in C grammar. like this:
1. define input_gadget_t struct
```c
// this struct is used to wrap the input functions, it can get data from va_list or from user defined functions
typedef struct {
    // get a long long
    long long (*get_long_long)();

    // get a long
    long (*get_long)();

    // get an int
    int (*get_int)();

    // get others
    ...

    va_list args;
} input_gadget_t;
```
2. change the ```vsnprintf_impl``` ```format_string_loop```function param type:
```c
// internal vsnprintf - used for implementing _all library functions
static int vsnprintf_impl(output_gadget_t* output, const char* format, input_gadget_t* input)
```
3. in the ```format_string_loop``` function, replace ```va_arg(args, int)``` to ```input->get_int()```, and so on.
```c
    else if (*format == '*') {
      const int w = input->get_int(); // va_arg(args, int);
      if (w < 0) {
        flags |= FLAGS_LEFT;    // reverse padding
        width = (printf_size_t)-w;
      }
      else {
        width = (printf_size_t)w;
      }
      ADVANCE_IN_FORMAT_STRING(format);
    }
```
4. and then the caller can define their own get_* function to pass their data, which may store in a array.

and I create a C++ library [printf_vector](https://github.com/esrrhs/printf_vector) to printf a vector, I had used this library code in my C++ library, so I just want to contribute a little to your library. so if the change above is not good enough, it's ok, just close it, because I found my C++ solution. 

thank you.

eyalroz, 2023-04-17:
I'm sorry, but that is well beyond the scope of a printf library. I'm glad you were able to adapt the code for your particular use.