eyalroz/printf #153: Rounding-Error

orgua, 2023-07-06:
Isn't that comparison (first line) wrong considering the following comment? 

```
    if ((!(remainder < 0.5) || (remainder > 0.5)) && (number_.integral & 1)) {
      // exactly 0.5 and ODD, then round up
      // 1.5 -> 2, but 2.5 -> 2
      ++number_.integral;
    }
```
[Link](https://github.com/eyalroz/printf/blob/master/src/printf/printf.c#L606)

I think it should be

```
    if (!((remainder < 0.5) || (remainder > 0.5)) && (number_.integral & 1)) {
```

eyalroz, 2023-07-09:
So, this is code from Marco Paland's original repository, which I haven't touched. It does seem a bit weird and I need to take another look.

orgua, 2023-07-09:
yeah, this is inherited from the original repo, but it's still wrong compared to the embedded comment. it came up during linting of a code-base.

Doing it like that with an extra step might fix some weird float-behavior with very small remainders in the fraction-part?!

eyalroz, 2023-07-09:
But - this is not what I have in the code right now... the code at the moment is:
```
  else if ((remainder == rounding_threshold) && ((number_.fractional == 0U) || (number_.fractional & 1U))) {
    // if halfway, round up if odd OR if last digit is 0
    ++number_.fractional;
  }
```
and that seems to match what the comment says.

eyalroz, 2023-07-09:
So, I just made it Banker's rounding.

HenryLin2000, 2023-11-24:
In fact, I found that when executing this block, there is always `remainder<=0.5`, because the `remainder` here is updated (in line 605)
https://github.com/eyalroz/printf/blob/f8ed5a9bd9fa8384430973465e94aa14c925872d/src/printf/printf.c#L604-L611
The `abs_number` here is the absolute value of the original number, and the value of `integral` depends on whether the number is carried.
There are 3 situations at this time
1. If the previous remainder>0.5 (carry), then the current `remainder` is a negative number
2. If the previous remainder<0.5 (no carry), then the current `remainder` is between the interval [0,0.5)
3. If the previous remainder=0.5 (special case), then the current `remainder` is also 0.5, and rounding processing is executed.

So, there is never `remainder>0.5`, and the condition here is really a bit strange.

commit 943989b0 (2024-07-25): Fixes #153: Now applying roll-over for values rounded up using Banker's rounding in `get_components()`

* Also added a comment-instead-of-a-testcase for a command in which that roll-over is meaningful
commit 2b6ece50 (2025-05-11): Fixes #153: Now applying roll-over for values rounded up using Banker's rounding in `get_components()`

* Also added a comment-instead-of-a-testcase for a command in which that roll-over is meaningful