eyalroz/printf #157: printf still bigger in size compared to stdio printf

ramadhanafif, 2023-07-13:
Hi, I was trying to use this library on the AVR ATmega328P microcontroller (Arduino). When comparing the file generated using stdio and this library, I noticed that the library resulted in a significantly larger file size. I believe I have set the option to use the minimum amount of memory. Here is the output of avr-size:
### using stdio.h
```
AVR Memory Usage
----------------
Device: atmega328p

Program:    1992 bytes (6.1% Full)
(.text + .data + .bootloader)

Data:         22 bytes (1.1% Full)
(.data + .bss + .noinit)
```
### using printf.h
```
AVR Memory Usage
----------------
Device: atmega328p

Program:    6942 bytes (21.2% Full)
(.text + .data + .bootloader)

Data:         34 bytes (1.7% Full)
(.data + .bss + .noinit)
```

and here the cmake options that I set
```
 ALIAS_STANDARD_FUNCTION_NAMES    NONE
 BUILD_STATIC_LIBRARY             ON
 BUILD_TESTS                      OFF
 CHECK_FOR_NUL_IN_FORMAT_SPECIF   OFF
 CMAKE_BUILD_TYPE
 CMAKE_INSTALL_PREFIX             /usr/local
 DEFAULT_FLOAT_PRECISION          6
 LOG10_TAYLOR_TERMS               4
 MAX_INTEGRAL_DIGITS_FOR_DECIMA   9
 PRINTF_DECIMAL_BUFFER_SIZE       32
 PRINTF_INTEGER_BUFFER_SIZE       32
 SUPPORT_DECIMAL_SPECIFIERS       OFF
 SUPPORT_EXPONENTIAL_SPECIFIERS   OFF
 SUPPORT_LONG_LONG                OFF
 SUPPORT_MSVC_STYLE_INTEGER_SPE   OFF
 SUPPORT_WRITEBACK_SPECIFIER      OFF
```

and this is where I code differs from stdio and printf library. Instead of the regular `vsnprintf` I use `vsnprintf_`
```
void myprintf(char *fmt, ...)
{
    char buf[128]; // resulting string limited to 128 chars

    va_list args;
    va_start(args, fmt);
    int len = vsnprintf_(buf, 128, fmt, args);
    va_end(args);

    for (int i = 0; i < len; i++)
        USART_Transmit(buf[i]);
}

```
the rest of the code is identical.

For build and link flags, I just use -Os to minimize code size.
Any tips on how to reduce the code size? Or is this library is better optimized for 32-bit mcu?

eyalroz, 2023-07-13:
Hello @ramadhanafif .

While I'm maintaining this repository - I do not do microcontroller development. So, I am not familiar with what happens with the AVR ATmega328P.

I would speculate that this is a "counting problem", i.e. that there is printf code in another object/executable/library, from the standard library. I am very doubtful the code size can decreased by a factor of 3. 

@phillipjohnston , perhaps you can chime in about this?

or, @ramadhanafif  - perhaps you could write [Phillip](https://github.com/phillipjohnston) and ask for his opinion on this?

ramadhanafif, 2023-07-13:
Thanks for the suggestion @eyalroz 

I do suspect that this is an unoptimized linker issues. Although I'm not sure how to do that

phillipjohnston, 2023-07-13:
In addition to @mickjc750's suggestion, You would also need -gc-sections as a linker flag.

For reference, with the original mpaland/printf library with similar formatting options disabled, I measured ~1.6kb on the same platform. I haven't taken similar measurements with the new library.

ramadhanafif, 2023-07-14:
@phillipjohnston avr-gcc is based on gcc 7.3, doesnt have `-gc-sections` flag
@mickjc750 Now I have added `-ffunction-sections` and `-fdata-sections` and `-flto` but it does not seem to change.

@eyalroz I found that the project cmake does not pass the compile or linker flags to the printf library. After checking the build commands, I discover that the libprintf.a are created without any optimizations. I modify the library CMakeLists and add a -Os. Here is the result

```
AVR Memory Usage
----------------
Device: atmega328p

Program:    3116 bytes (9.5% Full)
(.text + .data + .bootloader)

Data:         34 bytes (1.7% Full)
(.data + .bss + .noinit)
```

Definitely an improvement, but still somewhat bigger than the stdio printf (1992 bytes).
Will try to compile them manually instead of using cmake scripts, maybe we can find something else.

phillipjohnston, 2023-07-14:
Gc-sections is a linker flag. You need it for the others to make a difference.CheerPhillip

ramadhanafif, 2023-07-14:
@phillipjohnston Thanks for the reply.

I had tried using --gc-sections, but the avr-gcc returns error for not understanding what it means. 
Digging deeper into the manual, you can see that --gc-sections is not listed in the [avr-gcc man page](https://linux.die.net/man/1/avr-gcc)

ramadhanafif, 2023-07-14:
Hi @phillipjohnston , I found out what you mean. I have added the 
`-ffunction-sections -fdata-sections -Wl,--gc-sections`
but it size stays at 3116 bytes (no change).

eyalroz, 2023-07-14:
@ramadhanafif : I would accept a pull request which adds a CMake option for setting these compilation and linking options for the libprintf target.

Or, if you're not familiar enough with CMake to write that, I'll try to get to it soon.

@phillipjohnston : Using these flags, how far can the size be brought down on the AVR atmega328p?

eyalroz, 2023-07-23:
@ramadhanafif : ping.

ramadhanafif, 2023-07-24:
@eyalroz I don't know a good way to approach this issue. For the moment, I'm adding a compile flag to my main CMakeLists.txt 

```
add_subdirectory(printf)
target_compile_options(printf PUBLIC -Os)
```

eyalroz, 2023-11-10:
@ramadhanafif Would it not be enough to suggest the use of `-Os` in `CFLAGS` when configuring and building printf? In the README perhaps?

ramadhanafif, 2023-11-11:
I agree, that simplifies things

eyalroz, 2023-11-17:
@ramadhanafif : How's the README change I just introduced?

ramadhanafif, 2023-11-19:
That surely helps! Great job @eyalroz! thanks for solving this one until the end!

closed 2023-11-19 by ramadhanafif