eyalroz/printf #168: Incorrect size assumption for size_t on 16-bit target (msp430).

geurtv, 2023-11-20:
As noted in README.md:
> The implementation currently assumes each of intmax_t, signed size_t, and ptrdiff_t has the same size as long int or as long long int. If this is not the case for your platform, please open an issue.

With msp430-elf-gcc for the 16-bit msp430 microcontroller `sizeof(size_t) == 2`, which is the same as `int` and not as `long` (`sizeof(long) == 4`). A change like this should be portable and fixes the issue (`intmax_t` and `ptrdiff_t` will need similar changes):

``` c
flags |= sizeof(size_t) <= sizeof(int) ? FLAGS_INT : sizeof(size_t) == sizeof(long) ? FLAGS_LONG : FLAGS_LONG_LONG;
```

eyalroz, 2023-11-20:
So, you're saying `long` is not representable in a single machine register? Are you 100% sure this is the case? That's a very weird choice for a platform's C ABI.

geurtv, 2023-11-20:
Yes, that's exactly how it is, and I wouldn't expect this to be too uncommon for 16-bit microcontrollers. `int` (and `short`) will be the same size as the cpu register size, so 16-bit. Then `long` will be 32-bit and `long long` 64-bit (which gcc actually supports). For msp430 gcc has two addressing mode options: `-msmall` and `-mlarge`. With `-msmall` it uses 16-bit pointers and 16-bit `size_t`, with `-mlarge` it uses 20-bit pointers and 32-bit `size_t` (which should already work). The msp430 variant I'm using has 16k flash and 4k ram, so I use `-msmall`.

Btw, `intmax_t` already works as it's `long long`. For `-msmall` it's (only) `size_t` and `ptrdiff_t` that don't work.

eyalroz, 2023-11-20:
Ok, if you say so... I'll try to find the time to consider your suggestion and check the code for potential impact of this situation. 

However, as you may know, there's war now, and demonstrations against it, and I've already been arrested at one of those, and there are some personal issues, and my job etc., so it may take a bit of time. If I haven't done anything within... 3 weeks from now, please ping me.

geurtv, 2023-11-22:
That does the trick, thanks! Note that ptrdiff_t needs similar treatment.

eyalroz, 2023-11-22:
Right... can you recheck?

geurtv, 2023-11-23:
Yes, that now works for both `size_t` and `ptrdiff_t`.

commit 5e9b4daf (2023-11-21): Fixes #168: Now supporting the case of `size_t` being of smaller size than `long` (e.g. for the msp430-elf-gcc target)
commit 4046d01a (2023-11-22): Fixes #168: Now supporting the case of `size_t` and `ptrdiff_t` being of smaller size than `long` (e.g. for the msp430-elf-gcc target)
commit 641fe432 (2023-11-22): Fixes #168: Now supporting the case of `size_t` and `ptrdiff_t` being of smaller size than `long` (e.g. for the msp430-elf-gcc target)
closed 2024-07-19 by eyalroz