eyalroz/printf #173: Enable C89 support

lygstate, 2023-12-15:
For relative old but still usable OS such vxWorks 6.x, we want a cross-platform printf to handling printf things and use c89 syntax supported

eyalroz, 2023-12-15:
The library is currently written with C99 assumed. What would it take to make it compile as C89? Apart from the comment style change?

Also, do the C compiler on vxWorks 6.x really not support C99?

lygstate, 2023-12-15:
> The library is currently written with C99 assumed. What would it take to make it compile as C89? Apart from the comment style change?
> 
> Also, do the C compiler on vxWorks 6.x really not support C99?

its default option didn't support for double line comment

eyalroz, 2023-12-15:
> its default option didn't support for double line comment

And with a non-default option?

Also, there are many other changes which would be necessary to make the library C89-compliant. Specifically, we would need to not use `long long` and avoid mixing declarations and code. How come the vxWorks compiler accepts it then?

lygstate, 2023-12-15:
> And with a non-default option?
>
> Also, there are many other changes which would be necessary to make the library C89-compliant. Specifically, we would need to not use `long long` and avoid mixing [...]

Currently only use block line comment is enough. `long long` is supported

lygstate, 2023-12-15:
The header is present, it's just the header with `-ansi` option

eyalroz, 2023-12-15:
> Currently only use block line comment is enough. `long long` is supported

So, what you're asking is not C89 support, nor C90 support. Perhaps we can accommodate your compiler by some kind of CMakeLists.txt check, that would add an appropriate flag for `//` comment support?

lygstate, 2023-12-15:
> > Currently only use block line comment is enough. `long long` is supported
> 
> So, what you're asking is not C89 support, nor C90 support. Perhaps we can accommodate your compiler by some kind of CMakeLists.txt check, that would add an appropriate flag for `//` comment support?

Yeap, it's a -ansi option. I've seen a lot more compiling warning, I think achieve that would receive broader usage

eyalroz, 2023-12-17:
> Yeap, it's a -ansi option

I think you've misread what I've written. The `-ansi` option, at least with compilers such as clang and gcc, does not add support for anything; it is essentially equivalent to `-std=c90`, which contradicts std=c99, and restricts rather than relaxing constraints.

At any rate, it remains unclear to me what you are asking and for purpose.

lygstate, 2024-07-03:
> I think you've misread what I've written. The `-ansi` option, at least with compilers such as clang and gcc, does not add support for anything; it is essentially equivalent to `-std=c90`, which [...]

I am trying to use it within very old system such as VxWorks( and change the compiler option is not accepted) That's why I am trying to do c89 compat(Not allow `//`) comment

eyalroz, 2024-07-03:
I would consider a pull request which adds C89 support, if significant changes are #ifdef'ed ...

eyalroz, 2024-07-19:
There are other C90 limitations that the PR does not address.

lygstate, 2024-07-19:
> There are other C90 limitations that the PR does not address.

Yeap, if change all would be a large , thank you for working on this, I am busy on other things

eyalroz, 2025-08-25:
> is this issue still relevant or solved?

The state of this issue is still captured by this statement:

> I would consider a pull request which adds C89 support, if significant changes are #ifdef'ed ...

lygstate, 2025-08-26:
> TBH, I am doubtful that this would be a good idea, but I don't want to dismiss the possibility apriori. I'm personally fine with requiring C99 - a standard which by now is over a quarter-century old. But maybe I'm overblowing the detriment to convenience and readability.

With a new MR, all warning for C90 are addressed and enabled it with error to prevent future breakage