eyalroz/printf #180: Release Schedule

joshklod, 2024-07-09:
First of all, thank you for continuing to support this library.

I see that development has not halted, but there has not been a release in almost two years.  Why not?

I understand that new features should be thoroughly tested before being merged to master, but it would seem that at least a bugfix release is in order.  The `develop` branch appears to have several significant improvements over the latest release ([v6.1.0](https://github.com/eyalroz/printf/releases/tag/v6.1.0)).

Should the [v6.2.0-b1](https://github.com/eyalroz/printf/releases/tag/v6.2.0-b1) beta release be considered relatively stable, or is it subject to rebasing, like `develop`?

eyalroz, 2024-07-19:
Hello @joshklod .

There isn't a good reason for a lack of a release. The not-so-good reason is that I had waited for more issues to get fixed before completing the release, and it just slipped my mind.

eyalroz, 2024-07-19:
... new release is out.

joshklod, 2024-07-19:
Great! Thank you for your honest response and prompt action!

closed 2024-07-19 by joshklod