eyalroz/printf #184: "% .3g" failure: " 1000" instead of " 1e+03"

eyalroz, 2024-07-26:
```
fmt = "%+.4g" value = -9999.833007812500000000
libc:     "-1e+04"
our lib:  "-10000"

fmt = "% .3g" value = 999.779602050781250000
libc:     " 1e+03"
our lib:  " 1000"
```

ritesh006, 2025-08-26:
Could you please help me to reproduce the same issue

eyalroz, 2025-08-26:
@ritesh006 : First, thak you for taking an interest in the outstanding issues with this library.

To your point: Just run this program:
```
#include "printf.h"

int main() {
    printf_("ours:  % 01.1g\n", 9.8);
    return 0;
}
```
compiled with the library, of course.

ritesh006, 2025-08-28:
```
#include <stdio.h>
#include "printf.h"

void _putchar(char c) {
    putchar(c);   //
}

int main(void) {
    printf_("ours:  % 01.1g\n", 9.8);
    return 0;
}
```

```
ours:   10
````

is this the correct approach

eyalroz, 2025-08-28:
That program gets you the incorrect output that the library produces, yes.

eyalroz, 2025-11-04:
> what actually expected?

We should print: `1e+01`.

eyalroz, 2025-11-12:
Hmm... you may be write. This is odd...