eyalroz/printf #187: printf.h should also(only?) import printf_config.h

MrJake222, 2024-10-17:
For me, config file should be imported by `printf.h` in addition to or instead of `printf.c`.

For example, `printf.h` uses `PRINTF_ALIAS_STANDARD_FUNCTION_NAMES_SOFT` which can't be set by the config file currently.

eyalroz, 2024-10-17:
> `PRINTF_ALIAS_STANDARD_FUNCTION_NAMES_SOFT` ... can't be set by the config file currently.

It can be set using the config file.

But regardless of that - please argue you believe this should be the case, and which of the two options you're suggesting: Support in-addition-to or support instead-of.

MrJake222, 2024-12-11:
It's set to `0` in config [here](https://github.com/eyalroz/printf/blob/2d75fef6c605e699d2643267cbaeaaa49b0a2635/src/printf/printf.h#L66-L68) if not defined. And it won't be defined if not included.

ritesh006, 2025-08-26:
is this issue still relevant?