eyalroz/printf #192: GCC14 and 15 unhappy with putchar_

tomstorey, 2025-07-15:
Hi!

I have a toolchain targeted towards building code for the Motorola 68k family of CPUs, and I have your printf included as part of a "libc" like library.

Older versions of GCC seemed happy enough to compile your printf without any complaints, but I notice that on working to update my toolchain to build on ARM based Apple devices that GCC 14 and 15 are unhappy with the use of a call to `putchar_` and produce an error:

```
printf/printf.c:241:5: error: implicit declaration of function 'putchar_'; did you mean '_putchar'? [-Wimplicit-function-declaration]
  241 |     putchar_(character);
      |     ^~~~~~~~
      |     _putchar
make: *** [build/./printf/printf.c.o] Error 1
```
Changing `printf.c` to use a call to `_putchar` results in it compiling just fine again with GCC 14 and 15.

I have some vague recollection that this printf (maybe back before it was forked?) used to use `_putchar` but it was changed to `putchar_` for some reason that escapes my memory at this time.

Anyway, I just thought I'd bring this up as maybe a/the change needs to be (re)considered? Would appreciate your thoughts on this.

Thanks!

eyalroz, 2025-07-15:
Perhaps you neglected to include `printf.h`? It does declare `putchar_()`. The `_putchar()` function comes from the standard library implementation, and we cannot rely on its existence. _You_ are supposed to provide the `putchar_()` implementation - and it could very well simply be an invocation of `putchar_()`.

eyalroz, 2025-07-15:
So, I built this with GCC 14. No warnings, no trouble. I'm pretty sure you forgot to include the header.

tomstorey, 2025-07-16:
I am aware that I have to supply `putchar_()`, and the header file is most definitely included.

The only thing I had done differently was to update the version of GCC in my toolchain from e.g. 11.2.0 to 14.3.0 or 15.x (something that is happy to compile on more recent versions of macOS). The toolchain is several years old at this stage, with fresh installations on a number of my own machines at the least (and I do believe some other people have used it as well), so I would imagine that if I had simply forgotten something that it would have become apparent by now.

I am building this printf into a library that is then linked with other code that is built via another makefile in my repo. I have used it successfully in prior projects. But perhaps we have different compilation settings? This is the makefile that builds my library: https://github.com/tomstorey/m68k_bare_metal/blob/master/libmetal/Makefile

Worth noting, but my toolchain targets bare metal development with no underlying OS, so there are no standard libraries etc (as I understand it), and my library just provides a bunch of libc like stuff for familiarity sake. It seems to be an issue with what GCC thinks the call to `putchar` should be, but I dont have a deep enough understanding of this kind of thing to determine what the correct call should be. Either way it seems to want `_putchar` and wont let me build. Perhaps if you are building natively for your OS then it will be successful due to that function being available via included headers/libraries.

Perhaps using the `_putchar()` variant in my case is more appropriate? And I suppose my thought is whether this should be a configurable option for situations like this.

Toolchains and cross compilers arent my day job, this is a hobby project, and I am just trying to understand what I should be doing on my side is all.

eyalroz, 2025-07-16:
> I am building this printf into a library that is then linked with other code that is built via another makefile in my repo. I have used it successfully in prior projects. But perhaps we have [...]

What happens when you use CMake to build the library? Using the CMakeLists.txt available here?

tomstorey, 2025-07-19:
Using gcc 13.2.0 it builds fine in my library albeit with a warning about putchar_:

```
/mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-13.2.0/bin/m68k-eabi-elf-gcc -m68000 -Wall -g -static -I../include/ -msoft-float -MMD -MP -O3 -c -o build/./printf/printf.c.o printf/printf.c
printf/printf.c: In function ‘out_putchar’:
printf/printf.c:241:5: warning: implicit declaration of function ‘putchar_’; did you mean ‘_putchar’? [-Wimplicit-function-declaration]
  241 |     putchar_(character);
      |     ^~~~~~~~
      |     _putchar
```

Using gcc 14.3.0 it generates an error instead of a warning:

```
/mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-14.3.0/bin/m68k-eabi-elf-gcc -m68000 -Wall -g -static -I../include/ -msoft-float -MMD -MP -O3 -c -o build/./printf/printf.c.o printf/printf.c
printf/printf.c: In function ‘out_putchar’:
printf/printf.c:241:5: error: implicit declaration of function ‘putchar_’; did you mean ‘_putchar’? [-Wimplicit-function-declaration]
  241 |     putchar_(character);
      |     ^~~~~~~~
      |     _putchar
make: *** [Makefile:42: build/./printf/printf.c.o] Error 1
```

It would seem that for gcc 14+ they upgraded implicit function declarations from warnings to errors, and thats what causes it to break.

I don't have any prior experience with cmake, but heres what I got (doesn't look good). I first ran an `export CC=xxxx` to set the compiler I wanted to use then ran `cmake .`:

```$ cmake .
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-13.2.0/bin/m68k-eabi-elf-gcc
-- Check for working C compiler: /mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-13.2.0/bin/m68k-eabi-elf-gcc - broken
CMake Error at /snap/cmake/1468/share/cmake-4.0/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-13.2.0/bin/m68k-eabi-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/mnt/c/Projects/git/printf/CMakeFiles/CMakeScratch/TryCompile-BqjWLa'

    Run Build Command(s): /snap/cmake/1468/bin/cmake -E env VERBOSE=1 /usr/bin/gmake -f Makefile cmTC_822ce/fast
    /usr/bin/gmake  -f CMakeFiles/cmTC_822ce.dir/build.make CMakeFiles/cmTC_822ce.dir/build
    gmake[1]: Entering directory '/mnt/c/Projects/git/printf/CMakeFiles/CMakeScratch/TryCompile-BqjWLa'
    Building C object CMakeFiles/cmTC_822ce.dir/testCCompiler.c.o
    /mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-13.2.0/bin/m68k-eabi-elf-gcc    -o CMakeFiles/cmTC_822ce.dir/testCCompiler.c.o -c /mnt/c/Projects/git/printf/CMakeFiles/CMakeScratch/TryCompile-BqjWLa/testCCompiler.c
    Linking C executable cmTC_822ce
    /snap/cmake/1468/bin/cmake -E cmake_link_script CMakeFiles/cmTC_822ce.dir/link.txt --verbose=1
    /mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-13.2.0/lib/gcc/m68k-eabi-elf/13.2.0/../../../../m68k-eabi-elf/bin/ld: cannot find crtbegin.o: No such file or directory
    /mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-13.2.0/lib/gcc/m68k-eabi-elf/13.2.0/../../../../m68k-eabi-elf/bin/ld: cannot find -lgcc
    /mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-13.2.0/lib/gcc/m68k-eabi-elf/13.2.0/../../../../m68k-eabi-elf/bin/ld: cannot find -lc
    /mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-13.2.0/lib/gcc/m68k-eabi-elf/13.2.0/../../../../m68k-eabi-elf/bin/ld: cannot find -lgcc
    /mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-13.2.0/lib/gcc/m68k-eabi-elf/13.2.0/../../../../m68k-eabi-elf/bin/ld: cannot find crtend.o: No such file or directory
    collect2: error: ld returned 1 exit status
    /mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-13.2.0/bin/m68k-eabi-elf-gcc CMakeFiles/cmTC_822ce.dir/testCCompiler.c.o -o cmTC_822ce
    gmake[1]: *** [CMakeFiles/cmTC_822ce.dir/build.make:102: cmTC_822ce] Error 1
    gmake[1]: Leaving directory '/mnt/c/Projects/git/printf/CMakeFiles/CMakeScratch/TryCompile-BqjWLa'
    gmake: *** [Makefile:134: cmTC_822ce/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:5 (project)

-- Configuring incomplete, errors occurred!
```

I'm not really sure what Im supposed to do about this. It seems like it wants to build a program to run some tests and cant find some of the startup code that would be necessary to produce a binary to do that. But this wont be possible anyway as I have no way to execute code for Motorola 68k on my system.

The code produced by my toolchain would ordinarily be written to some ROMs and run in another machine with a 68k CPU, or I'd upload it to a 68k based machine with some of my own tooling and run it there.

eyalroz, 2025-07-19:
Seeing also the warning you get with GCC 13, this increases the certainty that your compiler search path is such, that it finds a different `printf.h` than mine.
```
/mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-13.2.0/bin/m68k-eabi-elf-gcc -m68000 -Wall -g -static -I../include/ -msoft-float -MMD -MP -O3 -c -o build/./printf/printf.c.o printf/printf.c
```
That  `../include/` is of course quite fishy, since my repository has no such directory. Perhaps there's another `printf.h`  which your compiler finds? That is something you should be able to determine when compiling with `-H`.

Otherwise - it could be a compiler bug I guess. It's also worth trying to manually-include the `printf.h` file, i.e. copy-paste it instead of the `#include <printf/printf.h>` directive, and seeing whether that works.

At this point I am resolving this bug, but we can continue the discussion.

----

About the CMake situation - you can invoke cmake with 
```
cmake -B /mnt/c/Projects/git/printf/build -S /mnt/c/Projects/git/printf/ -DCMAKE_C_COMPILER=/mnt/c/Projects/git/m68k_bare_metal/toolchain/m68k-eabi-elf-14.3.0/bin/m68k-eabi-elf-gcc
```
i.e. use that -D to set the C compiler. This is explained [here](https://cmake.org/cmake/help/latest/variable/CMAKE_LANG_COMPILER.html) (CMake is pretty well documented on their website, and there is StackOverflow with many relevant Q&A, plus you can ask here.)

tomstorey, 2025-07-19:
It is definitely picking up the expected printf.h - `../include` is a directory that contains all of the header files typically associated with the standard library, and I have printf.h in there.

That said, you gave me an idea, and I think I've figured out what has gone wrong, though I'm not sure why it has taken this long to become apparent...

At some point I have taken in a newer version of your printf.c, but had kept an older version of printf.h which is evidently prior to commit https://github.com/eyalroz/printf/commit/3f6d9725ab50906c107028b5c75b016adad2ce81

The version of printf.c I have uses a call to `putchar_` but printf.h still had reference to `_putchar`.

So I think I just need to update printf.h to match the release of printf.c (or maybe both since it is a couple of years out of date now) and I should be ok. With a little manual tweaking this now compiles just fine with gcc 14.